colloqi/piSignage #5: This player is powered by pisignage.com ticker

Magnekv, 2015-03-13:
Hi, uppdated to ver: 1.1.0/1.2.0  and now were getting a ticker saying ***this player is powered by pisignage.com*** how can we remove this?

ariemtech, 2015-03-13:
Hi, When there is no license file this promotional ticker is shown. If you give me your cpuids of piSignage players, I can send you a license files or upload them to pisignage.com. 
As part of the new release we are coming up with a local server which could control players, act as API gateway for our customers to add features, zoning support, HTML/URL/livestream support. We are planning to open source server and price the player software one time at USD25. Only difference between free and licensed version is the ticker.

Magnekv, 2015-03-13:
Hi, thank you very much for the quick reply and all the good information.  This looks very interesting!

Here are the cpu id:

00000000e5ab1a00

00000000bb025054

00000000e9229ac8

Yours sincerly
Magne Kvalsøren
Dagleg leiar | Konsulent
MK IT AS

Tlf. 412 97 643 | post@mkit.no<mailto:post@mkit.no>

ariemtech, 2015-03-13:
what is your installation/username at pisignage.com, I will upload them there.

Magnekv, 2015-03-13:
Mkitas

Login username : post@mkit.no

ariemtech, 2015-03-13:
Have uploaded the license files. If you reboot they should work. Otherwise you can download the files from URLs like http://pisignage.com/licenses/license_00000000e5ab1a00.txt

Magnekv, 2015-03-13:
Thanks, working good!

ariemtech, 2015-05-30:
Every installation comes with 3 free licenses(players). After which please purchase licenses at www.pisignage.com/lp to avoid "player is powered by pisignage.com" ticker

closed 2015-05-30 by ariemtech

jaderiverazxc, 2017-03-02:
HI. Im having the same problem about the ticker in offline mode. how can i remove?

navaneetharaopy, 2017-03-02:
Player needs to download license file from server. Before going to offline mode , make sure you register the player at pisignage.com and player comes online to download the license. After that it should work fine in offline mode also.

lu1smanu3l, 2021-01-02:
Hi.

If I buy a player license, how many player devices can I connect to the local server?

colloqi, 2021-01-03:
Hi

One, license is per player one time fee ($25)