colloqi/piSignage #6: Link view

PiXperience, 2015-04-22:
Hi,

I'm using the image from this link : 
https://dl.dropboxusercontent.com/u/166564018/pisignagepro.img.zip

I've got a problem, I cannot create a new Link.
When I click on Assets > Add > Add a link : it opens a popup with the parent window replicated into it.

G.

ariemtech, 2015-04-22:
Thanks for pointing the issue, one of the template files is missing for showing popup and hence the issue. Will fix the issue and release an update. Meanwhile you can use pisignage.com server to upload the link.

PiXperience, 2015-04-22:
Ok, I tried it, it send the link asset to the local Pi, but playing the playlist displays a Black Screen instead of the webpage...

ariemtech, 2015-04-22:
Can I see the link address? Also does it have flash content or heavy JavaScript, both could be tough for pi to handle.

PiXperience, 2015-04-22:
At the moment the link address just point a webpage with a simple text, I tried 9gag.com for example, but nothing better.
I already tried a previous version and webpages and assets management were working fine. I don't know why I've got this problem now.

ariemtech, 2015-04-22:
Thanks, will check and get back.

PiXperience, 2015-04-22:
I'll try with a previous pisignage.img I downloaded before. I'll try it tomorrow.

PiXperience, 2015-04-29:
Ok, I get the 'previous img' I already tested and had working. BUT it appears to be the SAME: "pisignage_113_130.img"...
So I don't know why this one is not working.
Now I can't synchronize playlists and assets from pisignage.com.

PiXperience, 2015-04-29:
I'm using another pi, and the synchronization is now ok, but I still have a black screen when displaying a link. And I cannot add a new link via local console.

ariemtech, 2015-04-29:
if you have access to pi through ssh, you can goto /home/pi/media directory and edit the ".link" directly.

We will release another image in next 1-2 weeks which should fix this issue (adding link via local console)

PiXperience, 2015-04-29:
Good to know, but it is stil displaying a black screen instead of the webpage. Pictures are working.

ariemtech, 2015-05-30:
With the 1.1.4 release this issue is fixed

closed 2015-05-30 by ariemtech