colloqi/piSignage #7: Unable to login!

explociti, 2015-06-19:
I entered 'pi' as the login and 'raspberry' as the password but apparently it they are incorrect login details. please can you tell me how to login.

navaneetharaopy, 2015-06-19:
use 'pi' for both login and password.

closed 2015-06-19 by navaneetharaopy

explociti, 2015-06-19:
Hi,
I have recently started using Pisignage and it perfectly suits my
requirements. During my use of Pisignage I noticed that the files that I
deploy are first downloaded onto the player/SDcard (not sure which) and
then they play on the screen. I want to know how to access and delete these
files from the player/SDcard.

regards,
Explociti

navaneetharaopy, 2015-06-19:
1.check in directory '/home/pi/media' (using SSH). 
2.visit  *'http://[pi ip]:8000/assets'* from Chrome browser (use EDIT button on top to delete files)

explociti, 2015-06-20:
Everytime I need to restart the player I have to reboot the system. And
reboot usually takes 3-4 mins. How to restart only the player?

ariemtech, 2015-06-20:
answered in zendesk