colloqi/piSignage #8: Make our own template for notice

Birssan, 2015-11-30:
Hello there,
Is there any way we can make our own template for notice ?
The default one has a max-width to 1000px which is kind of surprising considering we're using 1080p display, it should take the whole width don't you think ?

Anyway, at each update we have to override the t1_template.ejs so we can have our own template (different colors, blocks, etc)

Thanks

colloqi, 2015-11-30:
We will introduce the same like the one for welcome screen https://pisignage.com/lp/faq/#faq34.

Birssan, 2015-11-30:
Ok, but won't it be overwritten when we update piSignage ?

colloqi, 2015-11-30:
You need to upload the template to server or media folder and it will get downloaded after the update.

Birssan, 2015-11-30:
So you have to upload it as asset on the server ? (if the server is pisignage.com)

colloqi, 2015-11-30:
Yes

Birssan, 2015-11-30:
Ok thanks, looking forward to this :)

colloqi, 2015-12-23:
Addded support for this in the server and it will be there for player webUI in 1.4.0. 

Upload notice_template.ejs using Add Assets -> Upload files. Example is shown in https://github.com/colloqi/piSignage/blob/master/examples/templates/notice_template.ejs (based on the current template)

closed 2015-12-23 by colloqi