colloqi/piSignage #12: Portrait setup not play fullscreen video?

mtijerino78, 2016-01-28:
Hello,

When a portrait setup its set, video not playing at fullscreen.
 
When in Settings change of landscape to portrait and pi reboot images assets show in fullscreen but video 1080 x 1920 not, only fill until at center of screen.

but when video load splashscreen fill entire screen

thanks

navaneetharaopy, 2016-01-29:
1) What is your player version?
2) To which server your player is connected? Is it  pisigange-server(open source localhost) OR pisignage.com(cloud) ?
3) If possible attach some screen shots.

colloqi, 2016-01-29:
Hi 

We noticed the issue and it is coming because of fullscreen selection. Will correct in the next release.

Thanks for pointing out.

Regards
Ravi

note: you need to select the fullscreen option only in "2bp" and not needed in "2ap"

mtijerino78, 2016-01-29:
1) What is your player version?
omxplayer 0.3.6
2) To which server your player is connected? Is it pisigange-server(open source localhost) OR pisignage.com(cloud) ?
Conected to pisignage-serve(open source localhost)
3) If possible attach some screen shots.
I will send you a video soon.

mtijerino78, 2016-01-30:
Hello, 
I found this, when the setup its in portrait (display_rotate=1) omxplayer adjust the resolution to 1920 x 1080 like in landscape for me it would be that omxplayer adjust video to 1080 x 1920.

Thanks
[image]

colloqi, 2016-01-30:
Yes, actually when selecting "fullscreen", it is by default selecting layout "1" (instead layout "2ap" for portrait) dimensions. Will correct in the next release.

closed 2016-05-09 by colloqi