colloqi/piSignage #13: Resources are not deleted from the client

Birssan, 2016-02-02:
Hello,

I've noticed that when you delete a resource from the server interface, and remove them from the playlist, they're not deleted physically from the client (RPI), they're still in the media folder. Over time it could be problematic in matter of space... You have to remember to delete them once in a while from your media folder.

navaneetharaopy, 2016-02-02:
Thanks for the comment. This is in out to-do list. Also you can delete media files using shell command `rm /home/pi/media/< file-name> `.

Birssan, 2016-02-02:
Yep that's what I did, but I have to connect on my pi, and then remove files one by one, it's a bit tedious, plus I'm the only one who has access to the pi, other users are not really comfortable using shell commands, or anything that has to do with programming, etc...

nostahl, 2016-02-02:
I came here after looking for others with similar problem, I removed an item from the playlist on the server and saved changes (pressed done)   and it never cut it out of the playlist on the player.

Birssan, 2016-02-02:
You have to uncheck it from your playlist and then re deploy your playlist on the player

nostahl, 2016-02-02:
ok so I did that, now the video I removed from playlist on the local server and re-deployed it.  now its not showing on the pi  perfect.

I just logged into web interface of the pi and checked the playlist its playing locally and it still shows the video in that playlist...

colloqi, 2016-02-03:
Another way to clean up the media folder currently is to change the group of the player and change it back to the original group. When the group is changed the media directory is cleaned up.

colloqi, 2017-04-16:
Every deploy old files are deleted if not needed, fixed in a earlier 1.7.x release

closed 2017-04-16 by colloqi