colloqi/piSignage #15: PiSignage Local Server Not Working

GitRasp2, 2016-02-08:
-Colby

I am having trouble connecting to the local IP it gives me:
~ WIFI is up and running properly
~ DNS: Ok
~ PING: OK
~ Rebooted and reloaded multiple times

I had it all working about a week ago and now, It boots up just fine and then sits at the "Welcome!" splash screen, it is giving me the IP to connect to, but when I try to connect it tells me that the webpage is not available.  I am looking for any sort of help.

Thanks
--Colby

colloqi, 2016-02-09:
Colby

Everything looks fine, did you try the URL as http://<IP address>:8000 (notice port number 8000)?

regards
Ravi

GitRasp2, 2016-02-10:
Yes I did try that, and it still did not work.

I found out that it was a network problem though!  I fixed it and it is up and running!

Thanks for helping.

--Colby

closed 2016-02-10 by GitRasp2