colloqi/piSignage #16: Incorrect playlist duration when creating local playlist

lpipoz, 2016-03-09:
When using the local pi adress ip:8000
- uploading a video of more 300s
- creating a playlist with this video

the playlist restart after 5 min
the duration in the json is not correct

content of json:

```
{
    "name": "t3",
    "settings": {
        "ticker": {
            "enable": false
        },
        "ads": {
            "adPlaylist": false,
            "adInterval": 60
        }
    },
    "assets": [
        {
            "filename": "video.mp4",
            "duration": 10,
            "selected": true,
            "isVideo": true,
            "fullscreen": true
        }
    ],
    "layout": "1",
    "version": 1,
    "videoWindow": null
}
```

same json deployed via pisignage.com

```
{
    "name": "P9",
    "settings": {
        "ticker": {
            "enable": false
        },
        "ads": {
            "adPlaylist": false,
            "adInterval": 60
        }
    },
    "assets": [
        {
            "filename": "video.mp4",
            "duration": 2971,
            "selected": true,
            "isVideo": true,
            "fullscreen": true
        }
    ],
    "layout": "1",
    "version": 1,
    "videoWindow": null
}

```

GieVianto, 2016-05-09:
Hi, I have the same problem.

I was wondering when there will be a solution for this issue.

Thank you

colloqi, 2016-05-09:
Actually when the video is uploaded to the server it calculates the duration and fills in. In the next release (1.6.2 or 1.7.0) will add the video duration entry feature in playlist for videos uploaded directly to the pi.

Till then, if you are uploading video directly to the player, please edit the playlist file to modify the video duration from "10" to a number which represents the video duration in seconds (e.g. "duration": 3600,)

colloqi, 2016-05-20:
Please select the video duration in the playlist assets to reflect the video file duration (available in 1.6.2 onwards)

closed 2016-05-20 by colloqi

GieVianto, 2016-05-20:
Thank you,

By testing version 1.6.3, you can add the duration nicely, when activating the videofile.