colloqi/piSignage #17: Black screen between medias

st4rg33k, 2016-04-20:
Hi,
We have black screen between medias.

Hardware: Raspberry Pi 3
Server: Pisignage.com

Videos:
[IMG_1150.mp4.zip](https://github.com/colloqi/piSignage/files/228083/IMG_1150.mp4.zip)

colloqi, 2016-04-21:
Will look into it

navaneetharaopy, 2016-04-21:
Hi,

     Few things 
1) Animation was enabled and it was introducing a delay of 1 sec , but was not animating. We have disabled animation and now image transitions should be smooth after first round of caching.
2) Links will take time to load and there will be a gap.
3) In 1.6.0 beta we have a newer version of browser which you can give a try.

regards,
navaneeth

st4rg33k, 2016-04-22:
Ok. Thanks for your answer.
I will try this.

How install new version of browser ? sudo apt-get install...
The player is in production, i would not have to update all packages...

Regards,
Julien

colloqi, 2016-04-22:
Julien

By upgrading to new beta version in Software upgrade popup(in Players Screen by pressing orange cloud button) both the Software and Browser will be upgraded.

Regards
ravi

closed 2016-05-09 by colloqi