colloqi/piSignage #19: crash on 1080p

beankylla, 2016-05-19:
Hello,

Thanks a lot for the great software, I love it!

I just flashed 1.6.3 on to my 8 gig sd card + raspberry pi 2b.

whenever I put the resolution to 1080p, i get the first screen (with infos and so on) for a few seconds and then the Pi reboots into 720p.

Quite weird because I had 1080p running quite nicely on 1.6.1

the only thing in the error log is:

```
visit https://github.com/senchalabs/connect/wiki/Connect-3.0 for alternatives
connect.limit() will be removed in connect 3.0
connect.limit() will be removed in connect 3.0
connect.multipart() will be removed in connect 3.0
```
I don't really think itŝ related, is it?

Thanks and have a nice day,

colloqi, 2016-05-20:
The settings under Group > Display Set Mode takes precedence over the local player settings. So most likely it is going back to Group settings. Please change under Group settings to 1080p and it should work.

beankylla, 2016-05-20:
Hello,

This happens on a standalone pi not connected to any server. Are there group settings on standalone pis?

navaneetharaopy, 2016-05-20:
Do you have any ethernet or wifi connected to your player?

navaneetharaopy, 2016-05-20:
Your changes will be overwritten when you connect raspbery pi to internet.
So you can follow these steps:
## **Method1:** 
Just change the display configuration and disconnect ethernet/wifi connected to pi **(once player is online the settings will be overwritten)**
## **Method2:**
a) Register your player at pisignage.com.
b) Under **Deploy Groups > [player-group]**  click **set Display Mode** , select 1080p and deploy.
## **Method3:**
Following this method you may not be able to control your player from pisignage.com 
a) ssh to pi using **ssh pi@your-player-ip** (password: pi) 
b) open package.json , using  **vi /home/pi/piSignagePro/package.json**
c) remove  **pisignage.com** from **config_server** and **media_server**,  
Your package.json may look like this 
[image]

beankylla, 2016-05-20:
ok I got it, so even when I'm not connected to my pisignage account, once my pi reaches pisignage.com he automaticly takes the settings from pisignage.com and reboots?
is that it?

a bit weird that it happens but i guess there's a reason ^^

Thanks for the quick and precise answer anyways! :)

navaneetharaopy, 2016-05-22:
Yes, The cloud server pisignage.com has higher priority than local settings ( because if the player is deployed in remote place , user can deploy settings from pisignage.com) .

If you have a pisignage.com account you can control the player from any place. You get more features in pisignage.com compared to local player interface.

beankylla, 2016-05-22:
I'm aware of that, and it is a good functionality.

On the other end wouldn't be a little bit more logical to sync the settings only once I log into my pisignage account?

beankylla, 2016-06-03:
As far as i'm concerned this defect can be closed unless you want to change something related with my last comment :)

Thanks !

closed 2016-06-04 by colloqi