colloqi/piSignage #20: Translations

Birssan, 2016-05-25:
Hi there,

I've made some changes to the french translation file in February, no sign of activity from your part since then...
I can see now that the file is missing from the repository... 
Can you tell me what I have to do to propose french translations ?

colloqi, 2016-05-25:
Hi

We added a newer version of file which needs translation and hence deleted the old versions. Currently there is a French version. Is it not OK?

If you have suggestions please take the https://github.com/colloqi/piSignage/blob/master/translation-files/template.pot file and send us the translations.  We will update them. 

If you need the old translation file, please let us know.

We were updating the translation file and hence did not act on the February message.

Regards
Ravi

Birssan, 2016-05-25:
Hello,

Many translations are wrong...
If I understand well, I have to take the template file, create a new file and translate every word ?

You said there are new translations, it means you must have en `fr_FR.po` file right ? It's nowhere to be found on the repo.
IMO that would be faster to correct the current French translations

colloqi, 2016-05-25:
Hi

I just updated the  https://github.com/colloqi/piSignage/blob/master/translation-files/template.pot file, please take the new file.

Also uploaded the old french translation file. but the locations would have changed.

The current file is template.pot file for which old fr.po translations have to be plugged with corrections. Any help from our side for your translation, please let us know.

regards
Ravi

Birssan, 2016-05-25:
Ok thanks, I'll translate the strings when I can.

closed 2018-05-28 by colloqi