colloqi/piSignage #23: Network problem

st4rg33k, 2016-06-09:
Hi !

I have a problem...
In my network i have a TV Box for watching TV given by my operator. The TV Box use IGMPv2 for streaming diffusion.
When i connect the raspberry pi pisignage on the wifi network, the TV froze sometimes :(
I have increase bandwith TV but the problem can't change.

I have run tcpdump on raspberry pi and i see a weird exchange with the router.
The Rpi are 2 ip adress in arp table. Show the picture:

Router ARP Table
[image]
Wireshark:
[image]
[image]

Have you any idea what's happening ?
Thanks a lot !

colloqi, 2016-06-09:
Hi

In case of Static IP configuration we have noticed there will be two IP addresses assigned due to dhcpd which assigns a DHCP address also. But we did not face any issues because of that.

Are you using Static IP?

regards
Ravi

st4rg33k, 2016-06-09:
No, i use only DHCP wifi. The ethernet is disconnected.
[image]
[image]

colloqi, 2016-06-09:
With Jessie somethings have changed, can you change "dhcp" word to "manual" in 
a. iface eth0 inet dhcp
b. iface default inet dhcp

and try?

colloqi, 2017-04-16:
Closing the issue, let us know if it is still the issue?

closed 2017-04-16 by colloqi