colloqi/piSignage #25: How can i upgrade pisignage player?

beankylla, 2016-09-22:
From 1.6.1 or 1.6.3 to 1.7.2

beankylla, 2016-09-22:
I noticed the changelog was speaking about a version 1.7.9.

I downloaded what seemed like the last up to date version yersteday and it was 1.7.2.

Any way to upgrade to 1.7.9?

Thanks and have a nice day,

BK

navaneetharaopy, 2016-09-22:
Hi,
    Which Type of server you use? Is it open source OR cloud server (at pisignage.com)

a) If you are using open source you can restart the server and it will fetch the  new release files. And when the player is online click update button.
[image]

b) if your using cloud server(pisignage.com) just click update while player is online and player will be updated to the latest available version.

colloqi, 2017-04-16:
Closing the issue, let us know if it is still the issue

closed 2017-04-16 by colloqi

bcherup, 2020-03-20:
How do you manually upgrade ?