colloqi/piSignage #27: There is no date timer for publishing content

defender4github, 2016-10-13:
Dear,

I want to appreciate this project first. thank you for the work well done.
I observed that there is no date timer for publishing content. just like in cms when you can start the campaign and end it with respect to its contents.

can this be added?

colloqi, 2016-10-13:
Thanks

Under Group, you can add a playlist & schedule for a duration, or on particular week days/dates or a time duration in a day. Is something else you are talking about?
[image]

regards
Ravi

defender4github, 2016-10-13:
wow, this is awesome.
but I have not seen this in the local hosting server.
I have just installed it, so I am visiting it via my local ip address.

Please could you assist on how to go about it?

navaneetharaopy, 2016-10-13:
Hi,
 Which local hosting server your referring to ? 
Is it a) Open source server(port 3000) OR b) raspberry pi interface server(port 8000)?

defender4github, 2016-10-13:
raspberry port 8000

defender4github, 2016-10-13:
[image]

navaneetharaopy, 2016-10-13:
Hi,
Local player interface doesn't include this feature. For that you need to use either
1) https://pisignage.com/lp/
OR
2) https://github.com/colloqi/pisignage-server

defender4github, 2016-10-13:
hi navan, 
thank you for your prompt response, I had seen this link and i read the info there, but there was no HOWTO  as regards raspberry being used, ubuntu and others were listed.
So i wanna know, if pi can be used as the server as well as the player?
if yes, installation procedure is highly appreciated.
thank you

navaneetharaopy, 2016-10-13:
Hi,
Currently the open source server depends on other tools which require processing power. Since Raspberry pi is not designed to handle such processing,  Open source pisignage-server can't be hosted and it needs x86 hardware.

defender4github, 2016-10-13:
Hi,

Appreciated, creating Ubuntu server right now to handle that.

shall report back here

navaneetharaopy, 2016-10-13:
ok
Follow https://github.com/colloqi/pisignage-server#getting-started

defender4github, 2016-10-14:
hi
I have done the installation following the link above procedures, but i am stock at starting the server.js file.
error "txxxxn@signage:~/pisignage-server$ sudo node server.js
sudo: node: command not found

help pls

but when i issue -ls command,  the file is listed.

navaneetharaopy, 2016-10-14:
1) Which type of system you use ? (windows,mac,linux)
2) enter `node -v` in terminal,  the command should return the the nodejs version.
3) Please check `env` and `path` variable, These params should contain reference to nodejs source.
4) How did you install nodejs?

defender4github, 2016-10-14:
ubuntu 16.04 ---linux

navaneetharaopy, 2016-10-14:
ok, how did you install Nodejs ?

defender4github, 2016-10-14:
working now after installing legacy version of it again
but new error when i ran "sudo node server.js"

tgn@signage:~$ cd pisignage-server/
tgn@signage:~/pisignage-server$ sudo node server.js
   info  - socket.io started
Express server listening on port 3000 in development mode
*****************************************************
*     Unable to create media directory, exiting     *
*****************************************************

tgn@signage:~/pisignage-server$ 

help pls

defender4github, 2016-10-14:
oh, it is working now.

Thank you.

navaneetharaopy, 2016-10-14:
relative(at the same level) to pisignage-server create `media` folder. In order to store screenshots/thumbnails create a `_thumbnails` directory.

defender4github, 2016-10-14:
ok,
done, 

please can u guide on how to register my pi player on my local server?

thank u in advance

navaneetharaopy, 2016-10-14:
There is no registration involved. Either connecting usb keyboard(press Ctrl+n) to player OR http://< raspbery-pi-IP>:8000/settings change server to <your-server-address>:3000

defender4github, 2016-10-14:
Hi Navan,

thank you so much

defender4github, 2016-10-14:
Hi
In line with your last post, I am unable to connect to my local server pisignage with my pi player.
what is it i have not done correctly?

navaneetharaopy, 2016-10-14:
which step did you follow?

defender4github, 2016-10-14:
connecting usb keyboard(press Ctrl+n)

defender4github, 2016-10-14:
Is there a config file that i have to change the default 8000 port to 3000?

navaneetharaopy, 2016-10-14:
Press `Ctrl+n` and select `Config server` option, Change the the existing one with < your-pc-ip-address > :3000  and select ok

colloqi, 2016-10-14:
1.8000 is local port of player which need not be changed
2.it connects to local server over port 3000, so not changes needed
3. Just make sure under player settings server address is given for e.g. as 192.168.1.10:3000 where 192.168.1.10 is the *server* address
4. If still not connecting, see the log file /home/pi/forever_out.log which should say why it is not connecting

defender4github, 2016-10-14:
hi,

ok

i will try it once i am back 

thank you

defender4github, 2016-10-17:
hello

I have done it and it is working great.

thank you.

issue encountered
It does not start running at boot, i mean the server.js file

navaneetharaopy, 2016-10-17:
In ubuntu either you can enter and command in `~/.bash_profile` OR `~/.bashrc` OR  create a service file/ init script . So that your Nodejs open source server is live when system boots.

defender4github, 2016-10-24:
thank you done

closed 2016-10-24 by defender4github