colloqi/piSignage #28: Cannot sign in to local hosted pisignage server and player

defender4github, 2016-10-24:
In view of the subject matter, I want to reset my pisignage server password, any other means because i changed the  default password and now i can not log on again.

thank you in advance.

colloqi, 2016-10-24:
This should work

1. go to mongo shell (mongo)
2. use pisignage-dev or pisignage (database based on development or production flag)
3. db.settings.update({},{$set:{authCredentials:{user:"pi",password:"pi"}})

defender4github, 2016-10-24:
hi,
I have tried it and it is not working.
> show dbs 
replied with 
          admin (empty)
          local   0.76GB
          pisignage-dev (empty)

help please

colloqi, 2016-10-24:
Hi

Then server login should be pi & pi if the db is empty. are you talking about the player?

colloqi, 2016-10-24:
default htpasswd file to be put at /home/pi/piSignagePro has one line of content as follows

pi:$apr1$gVJMqj0q$YkBf8Z7hP2XzYFWTUHMEr1

defender4github, 2016-10-24:
Hello,
thank you so much for the guide. I was able to recover the password using mongo shell.
use pisignage-server
show collections
db.settings.find()

issue resolved.

closed 2016-10-24 by defender4github