colloqi/piSignage #29: Schedule multiple schedule display

JDL747, 2016-12-18:
Hi,

It would be nice to be able to set multiple screen schedules display ON/OFF for the whole week like so: 

**Monday**  : ON at : 09:00 - OFF at : 18:00
**Tuesday** : ON at : 11:00  - OFF at : 21:00

and so on ...

Are you planning to implement this functionality ?

Thanks

colloqi, 2016-12-18:
Hi

We have a TV_OFF playlist using which you can achieve this functionality. Schedule TV_OFF as the "default" playlist and schedule other playlists as you need using "add" button under Group details.

Whenever there is no additional playlist scheduled, default playlist(TV_OFF) will kick in and the TV will go off

Regards
Ravi

JDL747, 2016-12-20:
Thanks Ravi,

Works great !

closed 2016-12-20 by JDL747