colloqi/piSignage #33: Black Screens during transistion

abhishekricky, 2017-03-08:
Hi,

We are doing testing , we see when media is changed , there is appearance of black screen

Its looks a lot irritation 

Any Solution to it ?

navaneetharaopy, 2017-03-09:
Hi, can you try with latest pisignage image (https://s3.amazonaws.com/pisignage/pisignage-images/pisignage_1.8.5.img.zip) and let us know if the problem exists.

Thanks

abhishekricky, 2017-03-09:
let me try with it let u know today

abhishekricky, 2017-03-09:
> Hi, can you try with latest pisignage image (https://s3.amazonaws.com/
> pisignage/pisignage-images/pisignage_1.8.5.img.zip) and let us know if
> the problem exists.
>
> Thanks

sure let me try this today

abhishekricky, 2017-03-15:
Well its still coming black screens, tried new version 1.8.5

Spicefab, 2018-03-20:
Same problem with version 1.9.8c, any suggestion

zifnabbe, 2019-01-07:
I'm using software version 2.3.0 and see these black screens as well. Although not that annoying, is there a way to avoid them?

colloqi, 2019-01-07:
there is an option in group settings screen, please try that in 2.3.0

colloqi, 2019-01-08:
the gap between video -> video should go, if not can you send me the log "tail -200 /home/pi/forever_out.log" or check "which mpv" gives path of the mpv program?

zifnabbe, 2019-01-08:
How does the functionality exactly work? Is it a kind of preloading the next video? Could it also be useful for the issue with webpages which can take some time to load?

Concerning video -> video. I'll provide you the log through mail.

I see the following lines in the logging, just after the second video starts (after black screen)

```
8 Jan 22:08:37 - info: browser stderr: [789:2917:0108/220837.845164:ERROR:object_proxy.cc(626)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[789:2917:0108/220837.846728:ERROR:object_proxy.cc(626)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files

8 Jan 22:12:44 - info: livestreamer/ydl error WARNING: Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.

8 Jan 22:19:34 - info: livestreamer/ydl error WARNING: Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.

8 Jan 22:20:06 - info: livestreamer/ydl error WARNING: Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.
```

zifnabbe, 2019-01-11:
okay, thanks for the clarification. 
But, just to mention: I see an improvement (less black screens during transitions) even with steaming video assets (all my assets use streaming videos)