colloqi/piSignage #34: Multiple zones layout display issues

JDL747, 2017-03-09:
Hi,

It seems that there are some issues when using multiple zones layout and in my case, the `3a:  Three Zones(full bottom) with Main Zone on right': 

**1. Issue**: If I don't select `Stretch images to the screen` in the group settings, the image don't fit the space despite its resolution respect the zone dimension `(Side Zone: 320x540, Bottom Zone: 1280x180)`

[Screenshot](https://drive.google.com/file/d/0B4_yTbDkJlCaMEg2UVpiczBTRk0/view?usp=sharing)

**2. Issue**:  There is a delay of about 2s before the `side and bottom zones` appeared on the first cycle and it seems that there is box-shadow on the `side and bottom zones` which doesn't look nice.

[Video](https://drive.google.com/file/d/0B4_yTbDkJlCaRkl6NkhKVFVtblE/view?usp=sharing)

**3. Issue**:  After the first cycle, the `side and bottom zones` don't show the assets anymore. It's only a black zone.

[Video](https://drive.google.com/file/d/0B4_yTbDkJlCaQVNmMHdyM0Q5SEk/view?usp=sharing)

JDL747, 2017-03-10:
Any feedback will be more than welcome as it's a bit blocking right now.

Thanks

navaneetharaopy, 2017-03-10:
Hi, 

**For issue 1:**   
Please make sure your TV resolution is set at 720p(Since you have uploaded side images with dimension 320x540 ). 

Tv mode selection can be found under **Groups > Settings.**

[image]

[image]

**For issue 2 & Issue 3**
It would be great if you can attach a screenshot of your deployed playlist.

And Please provide details regrading
a) The type of raspberry pi used ( pi 3 , pi 2 OR pi b+ )
b) pisignage version used

JDL747, 2017-03-10:
Deployed playlist :

[image]

[image]

------------

Raspberry pi : Pi 3 Model B
Pisignage version : 1.8.6

colloqi, 2017-03-10:
Can I know your username at pisignage.com?

JDL747, 2017-03-10:
festitv16

JDL747, 2017-03-13:
Have you been able to find the issue?

navaneetharaopy, 2017-03-14:
Hi, currently we are not able to reproduce the issue in our player. We tested our player with similar playlist , our player played all the contents as expected.

closed 2017-04-15 by JDL747