colloqi/piSignage #36: Player error reporting

JDL747, 2017-04-20:
Hi,

Any idea why I'm getting the following error : 

[image]

The Group Settings is as follows : 

[image]

[image]

colloqi, 2017-04-20:
It is a harmless error, there are two places which can initiate a playlist start and if it is already started and it returns with a error. Will reduce the severity in next release.

JDL747, 2017-04-20:
OK, thanks !

I have a couple of questions regarding Scheduled Playlist: 

1.When setting TV_OFF as default playlist, do I need to check `Play together with scheduled playlist`?

2. On the `Set Timings for Scheduled Playlist` under `Only on Selected Days` if I select only 4 days out of 7, how do I set the dates?

[image]

colloqi, 2017-04-20:
1. It does not matter, but there is no need to select.
2. Let all the Dates be selected

closed 2018-05-28 by colloqi