colloqi/piSignage #37: Browser Language

gorbiWTF, 2017-04-21:
A webpage I want to display recognises the local language and is displaying the content in English rather than German. I set the language with "raspi-config" to German but that didn't have any effect so I guess it's the browsers language.

How can I change the language of Chromium through SSH?

colloqi, 2017-04-25:
Did you try the locale settings in Linux, I feel Chromium may be picking up the setting from Locale settings

gorbiWTF, 2017-04-26:
I changed the localisation to German with the command "raspi-config". When I have a "clean" Raspbian Jessie on the device and change it like this, Chromium is in German.

Is there any other way to change the language?

stevenbuehner, 2024-03-21:
When looking at the shell with `ps -ef | grep chromium-browse` it seems like chromium is opened with language disabled.

pi        3328  3270 40 00:43 ?        00:00:09 /usr/lib/chromium-browser/chromium-browser --type=renderer --crashpad-handler-pid=0 --enable-crash-reporter=,Built on Raspbian , running on Raspbian 12 --noerrdialogs --user-data-dir=/home/pi/apps-data/.config/chromium/weblink-2 --change-stack-guard-on-fork=enable --autoplay-policy=no-user-gesture-required --disable-notifications --force-device-scale-factor=1 --remote-debugging-port=0 --ozone-platform=wayland --lang=en-US --num-raster-threads=2 --enable-main-frame-before-activation --renderer-client-id=5 --time-ticks-at-unix-epoch=-1711063684981335 --launch-time-ticks=940632307 --shared-files=v8_context_snapshot_data:100 --field-trial-handle=0,i,5041670287349225139,6919032319776131382,262144 **--disable-features=TranslateUI** --variations-seed-version

I have no idea why this is and how to change it ... is there any progress up to now?

hitesh83, 2024-03-22:
**"--disable-features=TranslateUI"**  this flag is not to disable languages.
it is to disable a pop up in browser asking to convert non-english text to english. 
I am using Indian local languages with Unicode and with ttf fonts, both type works fine for us.

stevenbuehner, 2024-03-23:
> **"--disable-features=TranslateUI"** this flag is not to disable languages. it is to disable a pop up in browser asking to convert non-english text to english. I am using Indian local languages with Unicode and with ttf fonts, both type works fine for us.

Thanks for clarifying. Good to know. 

@hitesh83 Changing the localization with rasp-config and rebooting didn't work for me. It is still in English? How did you change your localization?

hitesh83, 2024-03-23:
I didn't change, Unicode/ TTF works without any change.

stevenbuehner, 2024-03-23:
Mhh ... maybe I need to specify more clearly my problem.
I have a website displayed with date time localization based on the browser language. I would like to display the website with a specific language (de_DE) but it is always renders in English. How can I tell the browser to use German localization?

stevenbuehner, 2024-03-23:
I pull back my question ... I found out that I am able to modify it with header request: Accept-Language: de-DE,de;q=0.9