colloqi/piSignage #39: Only Black Screen After Install

L30Yu, 2017-06-28:
Hi, I tried both of the method to set the raspberry pi on my Pi 3 model B. Both of them only give me a black screen. No Welcome Page.

The image i tried was the 1.9.0 based on Chromium-Browser

Then I tried reinstall the Raspbian with the latest RASPBIAN JESSIE WITH DESKTOP Image ( Version:June 2017 Release date:2017-06-21 Kernel version:4.9).
And followed the instruction changed the pi password, expanded file system, enabled SSH, selected "Boot to Desktop"

If you got a chance, could you try to install player and server from scratch and record a video with all details please.

colloqi, 2017-06-28:
Hi, the image 1.9.0 should work. The only issues we have seen so far of not booting (black screen) is due to
- SD card programming not proper or not properly inserted
- power supply not proper
- display type is not HDMI, in this case you need to chose proper hdmi group and type in /boot/config.txt (for which you need to use a TV in the beginning or ssh)

L30Yu, 2017-06-28:
You are right. After switched to a HDMI TV, this works. Thanks.

closed 2017-06-28 by L30Yu