colloqi/piSignage #42: Freeze screen when restarting Raspberry Pi !

JDL747, 2017-08-19:
Hi,

I'm having some issues when restarting my Raspberry Pi !

It always stops with "**Kernel Panic**" and the booting didn't continue!

[image]

Have you any ideas where it came from?

Thanks for your help !

navaneetharaopy, 2017-08-19:
Make sure you have SD card with class 10 rating. And the power requirement for raspberry pi is met 
     (**5v, 2A** https://www.raspberrypi.org/help/faqs/#powerReqs )
Do not plug power from your TV USB port , usually those ports have very less current (~500mA). The above problem could be error in reading a corrupted memory block , So just re-write the SD card with new pisigange [image](https://s3.amazonaws.com/pisignage/pisignage-images/pisignage_1.9.0.img.zip) and power ON.

JDL747, 2017-08-20:
The fact is, the raspberry pi was working correctly for a long time and it's just happened after a reboot and all the requirements mentioned are met.

closed 2017-10-26 by JDL747

andernunes, 2020-01-16:
Same problem.
Works fine for a time but when reboot i see the PANIC error and I need reflash and renew licence.
Sandisk Ultra 32GB
Oficial RPi power supply