colloqi/piSignage #43: Pisignage Access Database

ajaycrytek, 2017-10-01:
How to view the data in database. Required for study purposes. Please guide me to use any server management software required to view the changes we make from client side/Web Browser.

navaneetharaopy, 2017-10-03:
Hi,  You can access you local DB using mongo shell, which comes with set of standard commands. If you have issue with command line go for GUI based client such as [Compass](https://www.mongodb.com/products/compass) , [Robomongo](https://robomongo.org/)

ajaycrytek, 2017-10-03:
Hi navaneetha rao for replying But I am afraid I checked with all running services on RPI and there is no MongoD services running and those commands are ineffective. with my research i found its working with nodeJS and saving the data in json file. Correct me if I am wrong.

navaneetharaopy, 2017-10-04:
Hi , piSingage players (Raspberry pi software) **does not** include mongodb. All data saving is done through JSON file. you can refer **/home/pi/piSignagePro/config** directory which contains two JSON files named **cofnig.json** and **settings.json**.

closed 2018-05-28 by colloqi