colloqi/piSignage #46: TV On not working

bb4120, 2018-01-03:
Hi There,

I have two players that are having an issue with TV on in the morning. I have the players scheduled to go off at 9:00pm and on at 6:00am. 

The TV's go off correctly but they never come back on, and when i turn the tvs on manually the players aren't outputting anything to the TV so it goes to no signal and I have to reboot the players.

I'm fairly certain the TVs have CEC turned on since they are able to be powered off by the players, and the fact that they aren't outputting anything when i manually turn the tvs back on.

colloqi, 2018-01-05:
That indicates it is some issue with configuration (players aren't outputting anything to the TV) making player think still TV is OFF, can you let me know the username so that I can see the configuration?

bb4120, 2018-01-06:
Sure its medfordtv

colloqi, 2018-01-08:
The configuration looks OK, while TV turned on we use "tvservice --preferred;" which assumes your TV specifications are as per hdmi_mode and hdmi_group programmed in /boot/config.txt of Pi. Are you using standard 1080p TV (1 and 16)? if not, please program config.txt with your TV values.

colloqi, 2018-05-28:
Closing the issue

closed 2018-05-28 by colloqi