colloqi/piSignage #47: Playlist  deployment

adamongithub, 2018-01-22:
I built a player using Method 2, by installing Raspian and then installing pisignage on top. 

I've noticed that when I create a playlist on the server, upload an asset, configure the timings of assets and then deploy to player, the playlist is deployed but the timings don't appear to have an effect so the player sits on the Pending screen until I connect to the player locally and edit the playlist to assign time to an asset. 

I've also noticed that the Ticker config doesn't get sent to the player either. 

Is there something I'm doing wrong?

Thanks

adamongithub, 2018-01-22:
Hi, no I'm using the pisignage server at pisignage.com.

colloqi, 2018-01-23:
Can I know the user account so that I can have a look? You can write to us at support@pisignage.com

adamongithub, 2018-01-23:
Sure, it's tochinoki. Thanks.

colloqi, 2018-01-24:
Looks like /home/pi/media folder was stuck with a old __everyday.json file (playlist file), deleted the same and re-deployed.