colloqi/piSignage #51: Install CA certificate Trusted Root Certificate Authorities Store

glen4cindy, 2018-03-10:
I'm working on implementing PiSignage in a school on a network where all HTTPS traffic from devices must have the company CA certificate installed in the Trusted Root Certificate Authorities Store. I do this all the time on PC's and Mac's. I need help to get it into the Linux build that PiSignage runs on.

This is working on the network so far except for the YouTube links which are where the student news broadcasts are streamed from.

Can I get some help with this?

Thanks in advance.

ADSPL, 2018-03-10:
You may get some help from this link 
https://raspberrypi.stackexchange.com/questions/76419/entrusted-certificates-installation

I think PiSignage is using raspbian, so step mention in above link should guide you to install your CA certificate to Player.

glen4cindy, 2018-03-14:
Thanks ADSPL, I've been reading over this and it looks like it's what I need.

I am having a couple of issues.

It seems the only "shell" access available is from the PiSignage dashboard.

There is a >_ icon that says "Open player terminal to issue commands"

I tried using the "mkdir /usr/share/ca-certificates/local" directly but it didn't seem to work. I'll have to try again to get the exact error.

The 2nd problem I see is how I am going to get the cert from my source which is currently my flash drive to the "/usr/share/ca-certificates/local" directory. 

For the certificate to even work, it must be the one my network is looking for.

Thanks again.

ADSPL, 2018-03-14:
You can use shell from player dashboard.
Use sudo mkdir ...

You can connect your flash drive to Pi directly and use keyboard to open terminal. And install certificate.

If that is not possible, then zip the crt file, upload it in assets , add to desire player playlist and deploy it.

This will push your zip to players media dir.
/home/pi/media. , Use shell on dashboard and cp this zip to home dir.
Then remove this zip from playlist and deploy again, this will delete zip from media dir and won't give any trouble in display cycle.

After that just extract zip and cp crt to your desire location.

Use all shell commands with sudo

ADSPL, 2018-03-14:
Addition to above, 
If you are in same network as player,  you can directly SSH to player and do all the steps.

glen4cindy, 2018-03-15:
Hello,

I super appreciate the huge help you have been.

I figured out how to get into terminal on the TV the Pi is plugged into. CTRL-SH-T.

I successfully made:

sudo mkdir /media/usb1

I copied mycertname.cer into:

/usr/share/ca-certificates/share

I also tried copying it into:

/usr/share/ca-certificates/mozoilla

I also tried:

update-ca-certificates and dpkg-reconfigure ca-certificates

But my certificate was not listed.

(There is a specific name it has when installed. That name was not listed nor was the name of the .cer file I copied)

Then I attempted to reformat the certificate into PEM:

openssl x509 -inform PEM  -in mycertname.cer -outform PEM -out mycertname.cer

And received the following error:

Unable to load certificate

1995532384:error:0906D06C:PEM routines:PEM_read_bio:no start line:pem_lib.c:696:Expecting: TRUSTED CERTIFICATE

If you have another idea, I'd appreciate it.

Thanks again.

hitesh83, 2018-03-15:
there is one more folder for SSL certificates which uses PEM and CRT files  
path is **/etc/ssl/certs/** 
also this folder require symlink names of certificate.  (e.g.  ee1365c0.0 )
so need to refresh by **c_rehash** command.

  
```
sudo cp mycertname.crt /etc/ssl/certs/mycertname.crt
sudo c_rehash /etc/ssl/certs
```

glen4cindy, 2018-03-16:
hitesh83....

The Pi certainly does not like this certificate....

I continue to receive the "1995532384:error:0906D06C:PEM routines:PEM_read_bio:no start line:pem_lib.c:696:Expecting: TRUSTED CERTIFICATE" error.

I am seeing "Updating certificates in /etc/ssl/certs....unable to load certificate
1995532384:error:0906D06C:PEM routines:PEM_read_bio:no start line:pem_lib.c:696:Expecting: TRUSTED CERTIFICATE
unable to load certificate
1995532384:error:0906D06C:PEM routines:PEM_read_bio:no start line:pem_lib.c:696:Expecting: TRUSTED CERTIFICATE
WARNING: Skipping duplicate certificate mycertname.crt
1 added, 171 removed; done
Updating certificates in /etc/ssl/certs.... 0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d.....done."

I took the "*"'s off of all the listed certificates except the one I added. I left the ones on Thawte's because our network does require those.

I also read somewhere else that a .cer and a .crt is really the same so I changed the name of mycertname.cer to mycertname.crt to see if it would work differently but it didn't.

I'm also wondering if this message has anything to do with this issue. When we go to install this certificate on Windows machines we see this message "This CA Root certificate is not trusted. To enable trust, install this certificate in the Trusted Root Certification Authorities store."

Thanks again

hitesh83, 2018-03-16:
>  "This CA Root certificate is not trusted. To enable trust, install this certificate in the Trusted Root Certification Authorities store."

This shows your certificate is not signed by known provider, as it must be self generated. 
here is a good link to know how to get certificate sign and install it in Raspberry Pi, Ubuntu & Debian.

https://samhobbs.co.uk/2014/04/ssl-certificate-signing-cacert-raspberry-pi-ubuntu-debian

glen4cindy, 2018-03-16:
Thanks for the help, but, I'm not sure what this website suggests is possible.

The company I am trying to get this PiSignage player setup for is a school district.

At my level of technical support, I really do not have the information as to how the CA certificates are obtained or generated. We only know they are called "Akira".

Wheather or not that is a 'known provider' I really do not know.

Possibly there is an alternate way for these videos to play since this "https" website such as YouTube is may not be an option due to the restrictions our network has. :(

This has certainly been a fun learning experience and I appreciate all the help everyone has been. I've been learning quite a bit about Linux which I am liking too. :)

colloqi, 2019-04-20:
Closing the issue as it is more related to OS

closed 2019-04-20 by colloqi