colloqi/piSignage #52: Rasberry Pi 3b+

whitten2002, 2018-03-24:
Hi, has anybody got pisignage to work on a Raspberry pi 3b+ i'm only getting a rainbow screen

colloqi, 2018-03-26:
Looks like new Pi 3 B+ will boot only with the latest Raspbian (https://support.thepihut.com/hc/en-us/articles/360000553589-My-Raspberry-Pi-3B-won-t-boot-).

We are working on new Stretch based piSignage image and will release soon. Till then only option to try

if you have an older Pi 3B you can run "sudo apt-get update" followed by "sudo apt-get upgrade" on it... Then swap the MicroSD back into the 3B+
 

Also, note

Power Supply (PSU) - The new Raspberry Pi 3B+ needs more juice than previous models - You'll need an official 5.1V 2.5A power supply (or equivalent).

colloqi, 2018-05-21:
The latest image supports pi 3b+

closed 2018-05-21 by colloqi