colloqi/piSignage #56: Upgrade from 1.9.7d to ver: 1.9.9d not working

JDL747, 2018-05-15:
Upgrade doesn't seem to work despite the restart of pisignage and update progress screen.

After completion, the version stayed on 1.9.7 !

The upgrade was done from pisignage platform !

I would also like to point out that I'm having some random issues that take my screen offline and I would like to know if there is any monitoring network application that can be installed on the raspberry pi !

colloqi, 2018-05-21:
Are you still having this issue? let us know the account and player name at support@pisignage.com

colloqi, 2018-05-31:
Hello, Usually the error is indicated in one of the 3 files (mainly the first 2)
tail -200 /home/pi/forever.log - any error during upgrade
tail -200 /home/pi/forever_err.log - any error after upgrade till piSignage software start
tail -200 /home/pi/forever_out.log (logs of piSignage software)

please send us those files or let us know your username and player name by writing to  support@pisignage.com

JDL747, 2018-06-17:
Another issue when upgrading remotely !

[image]

hitesh83, 2018-06-18:
It looks like file system issue, or just SD card corrupt 
Need to re-flash with  latest version .
I had upgraded 1.9.7 to 1.9.9  to 2.0  on almost 80+ players and no issue at all.

colloqi, 2019-01-11:
Closing the issue

closed 2019-01-11 by colloqi