colloqi/piSignage #57: Volume Normalizer

skizztle, 2018-05-18:
Not sure if it could be possible on the Pi hardware but having a volume normalizer would be fantastic. We get a lot of our videos from different sources and the volume can be all of the place.

colloqi, 2018-05-21:
We tried looking at this once and then gave up, could be due to our expertise in this area

skizztle, 2018-05-21:
I would think handling it server side would make the most sense. Maybe run all videos through VLC on the server side after upload.

zifnabbe, 2019-01-07:
Noticed this aswell and was looking for a workaround.
Would the following alternative be a possible workaround/feature? A property on an asset (in the playlist), where you can manually configure the volume of the asset? 
This is ofcourse far from perfect (not automatic), but it would allow people to tweak it themself a bit

zifnabbe, 2019-01-07:
Thx for looking into it!
Correct. I understand the issue about more inputs :-). But a default input could be used (eg 100%), thus using the volume setting of the stream itself.

(or perhaps usage of a 'free text' parameter system. A kinda "raw" input method which is not required, ie empty)