colloqi/piSignage #58: Problems with upgrade to piimage 2.0.1

lewebsimple, 2018-06-12:
Hello,
I'm using the latest open source server and formatted the players using the 1.99aa image, which works just fine, then I'm trying to upgrade the players to the latest 2.0.1.

The first problem is when clicking the upgrade button next to the player in the list on the server, the 2.0.1 version is not shown in the line "1.99aa to ____"

The second problem is when clicking OK to start the upgrade, nothing happens. Upon inspecting the logs, it seems the player is looking for https://serverurl/releases/piimageundefined-v6.zip instead of piimage2.0.1.zip. I fixed this by creating a symlink on the server for the strange filename.

The last problem happens after the upgrade is complete, I can't log to the player on port 8000 with either the pi / pi default password or the one configured on the server. I tried updating piSignagePro/htpasswd manually and restarting the server, but it still does not recognize the password.

Can anybody point me in the right direction?

colloqi, 2018-06-13:
Hi, they all relate to the same, mostly package.json is not present at data/releases directory.

Can you pull the latest from pisignage-server repo and check the data/releases directory for package.json and other software image files? Also make sure your server can access pisignage.com to download these files.

lewebsimple, 2018-06-13:
data/releases/package.json is not in the repo, it is downloaded when launching the server (it did download fine on the server I'm having problems with). The same goes for the image files, they also get downloaded fine.

colloqi, 2018-06-14:
Still do you get "1.99aa to ____" , then please check the contents of package.json, version should be set to 2.0.1

lewebsimple, 2018-06-14:
This is the data/releases/package.json on the server:
```
{
    "name": "pisignage",
    "version": "2.0.1",
    "config_server":"pisignage.com",
    "media_server": "pisignage.com",
    "engines": {
        "node": ">=0.8.0"
    },
    "private": true,
    "dependencies": {
    	"apache-md5": "1.0.0",
        "express": "~3.4.3",
        "lodash": "~2.4.1",
        "ejs": "~1.0.0",
        "async": "~0.2.10",
        "node-uuid": "~1.4.1",
        "socket.io-client": "~0.9.16",
        "http-auth": "~2.1.7",
        "forever-monitor": "~1.2.3",
        "feedparser": "~1.1.3",
        "request": "~2.58.0",
        "robotjs": "^0.4.7",
        "ps-tree": "~1.1.0",
        "googleapis": "~1.0.7" ,
        "sanitize-html": "~1.11.1",
        "ws": "~1.1.1"
    }
}
```

colloqi, 2018-06-14:
Seems perfect, can I access the server via pisignage browser UI or ssh?