colloqi/piSignage #60: Hotter Raspi 3

cellx, 2018-09-27:
[image]

After updating to 2.1.7 my raspi got hotter. 80 degree. That is not normal. It has heatsink

ravibail, 2018-09-27:
Did you have a look at this? probably you can reprogram the SD card with new 2.1.0 image

1. https://help.pisignage.com/hc/en-us/articles/360015940932-Issue-with-2-1-0-image-

2. https://help.pisignage.com/hc/en-us/articles/360008078852-pi-3b-freeze-or-stop-working-after-few-hours

colloqi, 2019-01-11:
Assuming that the comments satisfied the question, else please write to support@pisignage.com

closed 2019-01-11 by colloqi