colloqi/piSignage #63: Rogue DHCP for de:ad:c0:de:ca:fe

bjhiggins, 2018-12-10:
Everytime I reboot one of my Pi-Signage Raspberry Pi's - all of my DHCP servers get a DHCPDISCOVER request for the MAC Address de:ad:c0:de:ca:fe

Information Security has flagged this as a problem.  

What can you tell me about this?

colloqi, 2018-12-13:
piSIgnage player acts as a DHCP client by default and sends DHCPDISCOVER message as per DHCP upon boot. Is there anything beyond this?

bjhiggins, 2018-12-13:
I see multiple DHCP requests from the Raspberry Pi.  One of the requests was made with the actual MAC Address of the Raspberry Pi.

The *other* requests use the MAC de:ad:c0:de:ca:fe.  This is not a real or valid MAC Address - and it reads "Dead Code Cafe".  This happens on at least three Raspberry Pi's I have in my immediate location.

The only thing I can find on this is https://help.fing.io/knowledge-base/getting-deadc0decafe-network/  It talks about a feature in the FING software which is a "DHCP server finder".

Does the piSignage image include some sort of "DHCP server finder" service?  If so - how can we turn it off.

colloqi, 2018-12-14:
Thanks for the hint, we use  nmap --script broadcast-dhcp-discover to find out the DHCP server and that does the similar thing (https://nmap.org/nsedoc/scripts/broadcast-dhcp-discover.html). You can modify the /home/pi/piSignagePro/misc/nwdiag.sh script to remove this and try

bjhiggins, 2018-12-17:
That worked.  For anyone watching, run this command:

sshpass -f FILEWITHYOURPASSWORD ssh -t -o StrictHostKeyChecking=no pi@YOUR.RASPBERRY.PI "sed -i -e 's/dhcpserver=/#dhcpserver=/g' piSignagePro/misc/nwdiag.sh"

Please create an RFE for putting an option in the piSignage server to disable this feature.  If your product is used in a corporate environment, I will guarantee InfoSec will flag and shut it down.

closed 2018-12-17 by bjhiggins

colloqi, 2018-12-18:
Thanks for the suggestion!

colloqi, 2019-02-11:
you can now disable it under user settings in server 
- do not show welcome screen and skip network diagnostics