colloqi/piSignage #65: [Sugestion] Advert asset for specific asset in main playlist

davidflypei, 2018-12-31:
We have a main playlist that plays through product videos. Right now we have web link assets that display price and info like that after the video. But we don't have a way to make a specific asset play directly after a specific main video when there's an advert playlist playing between every main video. There should be a way we can select a specific asset to play directly after a specific main video and then play the advert playlist and whatever is set after that. kind of like the sidebars except not on a sidebar and after the video.

colloqi, 2019-01-01:
Thanks for the suggestion, will explore implementation

davidflypei, 2019-07-20:
Any updates on this? I feel it would be an awesome and wanted feature to be able to play a specific asset directly after a main asset. Its also something we would really like before we look at deploying this to other stores.