colloqi/piSignage #67: [suggestion] (Un)Mute sound per asset (in playlist)

zifnabbe, 2019-01-07:
Hi,

Sometimes I would like to mute the sound of an asset (eg Video Stream), so the background music keeps playing. And sometimes I want to hear the sound embeded on a webpage (WebLink asset), so the background music is interrupted.

I found a way to handle this by playing with sound streaming assets. But it would be easier if you can configure this behavior through a property. 

Just a suggestion.

PS. It would be even nicer if we could fade in/out the background sound when switching between background and asset music. This would make the transition less abrupt. Further I've noticed that the background sound seems to stop and start quicker than the music of the asset. (I mean the background music stopped a few seconds before the asset music starts and starts a few seconds before the asset music has fully finished). Not a real issue, but I wonder if this can be tweaked somehow.

Regards,
Tom.

colloqi, 2019-01-08:
There is a "mute" option for every video asset under playlist

zifnabbe, 2019-01-08:
Sorry, I must overlook this. I can't find this "mute" option. I only see the option concerning the length

[image]

colloqi, 2019-01-11:
Mute is supported now only for video files and mute functionality is handled only for hardware assisted players (omxplayer and mpv). weblinks use chromium to play audio.

zifnabbe, 2019-01-11:
ok. Thanks for the clarification. As already mentioned, I found a workaround to achieve what I want.