colloqi/piSignage #68: TV is not starting anymore since version 2.3.0

zifnabbe, 2019-01-07:
I've a group with TV_OFF as default playlist.
And two scheduled play lists (morning and afternoon).

When playing the TV's go off as scheduled in the scheduled play lists.
But the TV doesn't activate again when the scheduled play lists are supposed to start.
This worked perfectly in the previous version. (I tested it with a previous version, stopped with testing a month or so and now started again with the new version. So I guess it is v2.3.0 related, but it also might be something else... although I can't think of something)

I can't activate the TV through the TV ON command (player menu). This worked before as well.
At the moment I can only activate it again by unplugging the power and re-plugging it in.

More info:
Furthermore, when playing, I can use the TV_OFF command (the TV switches off). Also, from that moment on, I can't use the TV_ON command to switch the TV back on. (again, for now, I can only activate it again through hard start of the pi).

I can access the pi trough ssh. So the pi is still active/reachable.

Any thoughts on how I can check/debug what is going wrong here?
Can I try out some ssh commands to turn it on from the command line?

Since TV_OFF is still working, I guess this is no CEC issue (or is my assumption wrong here?)

colloqi, 2019-01-08:
You could check the following
1. issue "echo scan | cec-client -s -d  1" command from ssh and see whether TV is reported (there should be two devices listed). If not pi is not able to detect the TV through CEC - check the TV settings for enabling CEC and cable 
2. There should be a gap of 1 minute to issue TV on after TV off is issued manually
3. Share the log "tail -200 /home/pi/forever_out.log" (support@pisignage.com) so that we can have a look

I believe it is not a 2.3.0 issue

zifnabbe, 2019-01-08:
Weird...
First things first. It started to work... but I don't know why. Here is more info.

I saw the info concerning 1 minute gap on your support site, so I knew this. I tried after a longer gap. So this is not the issue.

"echo scan | cec-client -s -d 1" always gives me the following output.

With the 'old' cable, a 'new' cable.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
pi@raspberrypi:~$ echo scan | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
~~~~~~~~~~~~~~~~~~~~~~~~~~~

So not two devices.

However, after I connected the pi to the TV with the new cable. I was able to put the TV OFF with the player command (was also possible earlier), but I also could put the TV back ON, through the command. 
I then replaced the cable by the 'previous' one. This time I can also put the TV back ON... so it is not the cable.

the TV now also starts through the scheduling....

So I don't know what is/was causing the issue  

Further:
the forever_out.log gives an info line 'cec is NOT supported'.  But it seems that piSignage can put the TV ON and OFF... When using the player TV ON / OFF commands, I see the following in the logs:

8 Jan 21:27:44 - info: openvg_display arguments: 1 2 0 0
8 Jan 21:27:44 - info:  **** turning tv off *****
8 Jan 21:27:49 - error: openvg_error: Failed to parse input: -2
Input was: 801

8 Jan 21:28:14 - info:  **** tvservice off *****
8 Jan 21:29:15 - info:  **** turning tv on *****
8 Jan 21:29:45 - info:  **** cec-client on *****
8 Jan 21:29:45 - info: openvg_display arguments: 1 2 0 0

Any ideas what might have caused the issue and why it does work at the moment?

(I will also send a log of the last 200 lines to support@pisignage.com)

zifnabbe, 2019-01-09:
This is really weird. It's again not working. I'll send the logs, but last input there is from yesterday 23:03...

zifnabbe, 2019-01-11:
Hmm, after further analyzing, couldn't this be the same issue as mentioned in issue "Player Offline #66"

colloqi, 2019-01-11:
I just configured TV ON at 12:15 
[image]
on a pi 3 running 2.3.0 with a TV which supports CEC. Here is the sample log
11 Jan 12:10:31 - info: *** sync initiated at server: 1/11/2019, 12:10:30 PM
11 Jan 12:10:31 - info: Spawning wget -rNnHd,--no-parent,-t,20,--waitretry,60,--retry-connrefused,--no-cache,-R,Test Group,pisignage.com/sync_folders/e2etest/Test Group; Running as PID 10291
11 Jan 12:10:32 - info: wget stopped with code 0 and signal null; downloadInProgressCount: 1
11 Jan 12:10:32 - info: COPIED SERVER FILES Test Group SUCCESSFULLY.
11 Jan 12:10:32 - info:  **** turning tv off *****
11 Jan 12:10:32 - info: Change playlist,deleting files: Intro40sec.m4v,Test Group.tmp,__video test 3b.json
11 Jan 12:11:02 - info:  **** tvservice off *****
11 Jan 12:15:00 - info: Schedule is changed to playlist short videos
11 Jan 12:15:00 - info:  **** turning tv on *****
11 Jan 12:15:00 - warn: *** skipping callback: 181,182
11 Jan 12:15:00 - info: *** Changing Playlist *** to short videos
11 Jan 12:15:00 - info: merged videos for play: /home/pi/media/Chefs Drink 5 Sec FINAL.mp4,/home/pi/media/Christmas Carol Menu Horz TV Slide D.mp4,/home/pi/media/Main.mp4
11 Jan 12:15:00 - info: Chrome loaded page /home/pi/piSignagePro/templates/1.html;?1547189100486
11 Jan 12:15:30 - info:  **** cec-client on *****
11 Jan 12:15:30 - info: openvg_display arguments: 1 2 0 1
11 Jan 12:15:30 - error: Already playing started

colloqi, 2019-01-11:
TV ON/OFF functionality works as follows. 

There are two events happen 
1.  one switching TV ON/OFF using CEC commands OR executing a python script in case RS232 controlled TV sets (script need to be written and stored in /home/pi directory TVon.py and TVoff.py)
2. Switching off the HDMI interface on the player 

Now during TV OFF
- CEC command is sent to TV to switch off which most TVs ignore
- After 30 seconds, HDMI interface on player is switched OFF - TV reports "No Signal" and typically switches off itself after a certain timeout. So this function works even if CEC is not supported

during TV ON,
- HDMI interface is switched ON on player , now TV won't switch ON
- After 30 seconds, CEC command to TV is sent. If this is ignored or CEC not supported. So TV will not be ON

In summary, if the above mentioned logs are reported in log file and TV does not switch ON, it is mainly due to HDMI cable or the TV not supporting CEC functionality. Also make sure in the TV Remote control menu CEC (may be called by different name) is enabled.

In any case, let us know at support@pisignage.com if you are facing issues not mentioned here

colloqi, 2019-01-11:
Commands to check whether CEC is working

echo scan | cec-client -s -d  1  (should display both TV and player devices)
echo help | cec-client -s -d  1 (help command)
echo “on 0" | cec-client -s -d  1 (CEC command to switch ON TV)
echo “standby 0" | cec-client -s -d  1 (CEC command to switch OFF TV)

Also, in PI there is a tvservice command . Few examples
 
tvservice -s (shows TV settings)
tvservice -o (off)
tvservice —preferred (switch on as per /boot/config.txt values)
tvservice —explicit=“CEA 4 HDMI”

zifnabbe, 2019-01-11:
Hi just to recapitulate the experience. Because during the latest test, it is not really an issue that the TV does not switch ON anymore. It does switch ON, but after a certain time of inactivity, it doesn't do that anymore.

I'll send detailed logs and real example detail later on:

A schedule 07:45 - 13:00 o' clock
A schedule 16:00 - 20:30 o' clock
A schedule 21:00 - 21:30 o' clock

Day1: 
To start the player, I unplug and replug the power around eg 17:00h.
The player starts and TV shows presentation
The TV stops playing at 20:30h.
The TV starts again playing at 21:00h
The TV stops playing at 21:30h

Day2:
Nothing happens, while the TV should start at 07:45
Nothing happens at 16:00
Nothing happens at 21:00

...

When trying to execute a command from the piSignage website GUI to the player. You get a "Could not get response from the player,Make sure player is online and try again."

During all that time the SSH tunnel to the player is still active. So there is still connection on network level. 
The logs don't show anything on Day 2.

So it seems that after some inactivity, the player 'seems' to become offline for the "piSignage website GUI". 
I guess therefor also no scheduling is performed on Day 2.

How can I revert to the previous version of piSignage? (Do I then loose the existing playlists, etc? Or can they be migrated back to the previous version) 
This way I can check this behavior in the previous version. (I didn't experience this in that version, so I would like to check this again)

zifnabbe, 2019-01-12:
thanks for the new commands. I've noticed the following.

Started the player by power on/off (removing power cable)
TV is activated and starts playing.
TV is deactivated at 13.00 o' clock
TV doesn't start at 16.00 o' clock as it should

SSH tunnel is still connected. piSignage gives message that player is offline
Last log message is at 13:23:57 --> so no commands are giving to start up at 16.00 o'clock.

Through SSH connection, excecuting following commands:
tvservice -s --> state 0x120002 [TV is off]
tvservice --preferred --> Powering on HDMI with preferred settings

The TV switches on. Sound is provided by the audio streaming asset of the playlist.
But nothing is shown! Only audo stream is active

tvservice -s --> state 0x12000a [HDMI CEA (4) RGB lim 16:9], 1280x720 @ 60.00Hz, progressive
tvservice -o --> TV switches off

colloqi, 2019-01-15:
- After the command tvservice --preferred, for the display to appaer you need to issue "sudo chvt 6; sudo chvt 7" commands 
- We are going to test a player with the following schedule and 2.3.0 and test here
A schedule 07:45 - 13:00 o' clock
A schedule 16:00 - 20:30 o' clock
A schedule 21:00 - 21:30 o' clock
- If you could setup the player with the problematic schedules and keep online, we could analyse the issue. Some of the weblinks which need heavy JavaScript/flash processing may stall the processor and keep player from coming online. Otherwise under any scheduling conditions player should be online

zifnabbe, 2019-01-17:
Hi, 

Thanks for looking into this.

I've done some testing with different configurations. I think I found the issue... 
The player has been up for two successive days. TV was scheduled as it should.
Today it doesn't start anymore.
For some reason I was starting to think that the issue is caused by the ticker (group level).

So the only difference is that yesterday I again enabled the group ticker. And today the TV doesn't start when it should.

What I found so far:

Scheduling of the TV works when **group ticker is disabled**.
Scheduling of the TV does not work (or works when time between on/off scheduling is short) when the **group ticker is enabled**.

I'll send the logs I've gathered later on by mail.

I still have to check if we get the same behavior when enabling the ticker on playlist level.

colloqi, 2019-01-17:
Thanks, Can you send us a screenshot of the group ticker configuration popup?

zifnabbe, 2019-01-18:
The same problem when the ticker is enabled on playlist level.
After disabling the ticker on playlist level (and non enabled on group level), the schedules work as expected.

Here are the screenshots of the ticker(s)

Group ticker
[image]

Playlist ticker
[image]

Just for completeness:
When the player does not wake up the tv (and appears to be offline from the webgui pisignage.com)
"tvservice --preferred", activates the tv
after this command, using "sudo chvt 6; sudo chvt 7" nothing happens

kd8ssq, 2019-02-03:
I'm also experiencing the same issue.  I just updated my software this week to 2.3.0 and when the schedule came to turn the TV on this morning, nothing happened.  The pisignage website showed a status of Online and running my playlist this morning.

colloqi, 2019-02-04:
Can I know the username at pisignage.com? You could also write at support@pisignage.com

zifnabbe, 2019-02-04:
Hi, 

@kd8ssq : are you using tickers? 

@colloqi :
Are you able to reproduce the issue with the tickers? 

Now, I recently noticed the player is telling there is a new version available: 2.3.0...  So somehow it must have 'downgraded' it somehow. Unfortunately, I'm unsure when this happened (before or after the tests). Anyhow, If before the test, then the problem exists in the previous version as well. At the moment the player is operating without the tickers enabled and everything works as expected (unfortunately, without RSS news feed at the moment)
At the moment I'm not upgrading to 2.3.0 I'm planning to buy an extra pi to try out first.

Regards,
Tom.

kd8ssq, 2019-02-04:
No, I'm not using any tickers

kd8ssq, 2019-02-04:
> Can I know the username at pisignage.com? You could also write at [support@pisignage.com](mailto:support@pisignage.com)

adelay

colloqi, 2019-02-05:
@kd8ssq , from the logs it looks like player send CEC command which was not taken by TV. So player kept reporting that TV is OFF. TV OFF works because after the CEC command, the HDMI is switched off. then TV reports no signal and may switch off depending on TV.

When the TV does not turn on, can you try sending 

echo “on 0" | cec-client -s -d  1 and see if it turns on.

3 Feb 17:26:00 - info: Schedule is changed to playlist Missions
3 Feb 17:26:00 - info:  **** turning tv on *****
3 Feb 17:26:00 - info: *** Changing Playlist *** to Missions
...
3 Feb 17:26:30 - info:  **** cec-client on *****
...

3 Feb 17:30:36 - warn: CEC status different than desired state, true;false;power status: unknown
;undefined
3 Feb 17:35:36 - warn: CEC status different than desired state, true;false;power status: unknown
;undefined
3 Feb 17:40:36 - warn: CEC status different than desired state, true;false;power status: unknown
;undefined
...

3 Feb 19:55:37 - warn: CEC status different than desired state, true;false;power status: unknown
;undefined
3 Feb 20:00:00 - info: Schedule is changed to playlist TV_OFF
3 Feb 20:00:00 - info:  **** turning tv off *****

colloqi, 2019-02-05:
@zifnabbe , unfortunately we could not reproduce. Currently 2.3.1 is available and hence upgrade button may be showing.
If you setup for a day with ticker and send me the log (tail -200 /home/pi/forever_out.log) it will help in analysing.

kd8ssq, 2019-02-05:
Thanks Ravi.  I'll give that a try.

zifnabbe, 2019-02-21:
With the new raspberry pi setup I indeed can't reproduce the issue (after solving the hostname issues #74). 
I'm now going to retrieve the other setup to perform new tests.

closed 2020-01-05 by colloqi