colloqi/piSignage #70: Setting the clock + new wifi

ElLocoFrances, 2019-01-31:
Hello everyone,

First, thanks for this amazing stuff ! Then I would like to know several things : 

- Can I switch my wifi access and keeping the licence ? Is the licence proper to one specific connection or to the player ?
- How can I set the clock so that it works all the time even if offline ? I've tried with a DS3231 but could not set it up. 
- Is it possible to put the layer online one time, then cut the connection and leaving an USB key with the group loaded ? The aim is still to keep the cock right. 

Thanks for your anwers

colloqi, 2019-02-01:
Thanks!
- License is tied to CPU ID of the player and downloaded as a file from the server after which it does not need to be online
- It should work, piSignage is based on Raspbian OS and you could set it as per OS instructions
-The player downloads the files (videos, images, zip, pdf, html, text) to the SD card and plays from there, so player need not have to be online once the files are downloaded

ElLocoFrances, 2019-02-01:
Thank you for your answers :) 

How do I open a terminal while player is working ? CtrlAltT does not work !

colloqi, 2019-02-01:
Video always plays on top of terminal, if video is playing, you may have to blindly type "sudo pkill omx" to kill the video to see terminal (after pressing ctrl+alt+t)

ElLocoFrances, 2019-02-01:
Great, but it appears just for one second, is that normal ?

hitesh83, 2019-02-01:
use any other system in same network and ssh to player, it will be easy.

colloqi, 2020-01-05:
Another option is first press ctrl+N and then ctrl+alt+T

closed 2020-01-05 by colloqi