colloqi/piSignage #73: piSignage interfering with a service

dexluce, 2019-02-15:
Hello,

I use a PIco HAT (https://www.modmypi.com/raspberry-pi/power-1051/ups-boards-1051/ups-pico) on my pi.
The HAT use a service booted at startup. The service is started well. But as soon as the piSignage service boot (as soon as the screen with info about the player is loaded), the HAT service crash without logs.

Do you have any idea what could crashed the HAT service?

Thank you

colloqi, 2019-02-16:
Can you provide more info about the HAT service? the /home/pi/start.sh in Pi is where pi services are started. You could comment those lines and restart to see who stops the service in manual mode.

dexluce, 2019-02-18:
I don't know much about the HAT service. It use python and GPIO python lib to monitore and control the HAT.
I opened an issue at PIco HAT, it gives more info about the HAT: https://github.com/modmypi/PiModules/issues/38

And I tried testing the /home/pi/start.sh line by line. It appear it is this line who somehow send a SIGTERM to the HAT service:
```
node pi-monitor.js
```

So I commented this line and all lines after. Rebooted the pi, and tried to launch ```node pi-monitor.js``` manualy to access logs. The result:
```
pi@pi-image-player:~/piSignagePro$ node pi-monitor.js 
Ignoring forever log trim error: Error: Command failed: wc -l /home/pi/forever.log
wc: /home/pi/forever.log: No such file or directory

{ lo: 
   [ { address: '127.0.0.1',
       netmask: '255.0.0.0',
       family: 'IPv4',
       mac: '00:00:00:00:00:00',
       internal: true },
     { address: '::1',
       netmask: 'ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff',
       family: 'IPv6',
       mac: '00:00:00:00:00:00',
       scopeid: 0,
       internal: true } ],
  wlan0: 
   [ { address: '192.168.1.95',
       netmask: '255.255.255.0',
       family: 'IPv4',
       mac: 'b8:27:eb:ed:29:02',
       internal: false },
     { address: 'fe80::59df:27cb:c181:fef0',
       netmask: 'ffff:ffff:ffff:ffff::',
       family: 'IPv6',
       mac: 'b8:27:eb:ed:29:02',
       scopeid: 3,
       internal: false } ] }
connect.limit() will be removed in connect 3.0
connect.multipart() will be removed in connect 3.0
visit https://github.com/senchalabs/connect/wiki/Connect-3.0 for alternatives
connect.limit() will be removed in connect 3.0
Express server listening on port 8000
18 Feb 12:21:46 - info: resolution changed to 1080p
18 Feb 12:21:46 - info: orientation set to : landscape
18 Feb 12:21:46 - info: Spawning wget -rNnHd,--no-parent,-t,20,--waitretry,60,--retry-connrefused,--no-cache,-R,Horizon_Group,pisignage.tree-it.ch/sync_folders/dexluce/Horizon_Group; Running as PID 980
18 Feb 12:21:46 - info: Chromium is available at /usr/bin/chromium-browser

18 Feb 12:21:46 - info: Starting websocket
18 Feb 12:21:46 - info: Openvg is available at /usr/local/bin/openvg_display

18 Feb 12:21:46 - info: evince pdf reader is available at /usr/bin/evince

18 Feb 12:21:46 - info: wget stopped with code 0 and signal null; downloadInProgressCount: 1
18 Feb 12:21:46 - info: Spawned openvg_display process: 1020 -> 1,0,0
18 Feb 12:21:46 - info: openvg_display arguments: 1 0 0
18 Feb 12:21:46 - info: COPIED SERVER FILES Horizon_Group SUCCESSFULLY.
18 Feb 12:21:47 - info: Browser loading /home/pi/piSignagePro/templates/1.html; Running as PID 1078
18 Feb 12:21:47 - info: ces support: opening a connection to the CEC adapter...
CEC version unknown
;;null
18 Feb 12:21:47 - info: cec is NOT supported
18 Feb 12:21:51 - info: browser stderr: Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any effect anymore. please remove it from your fonts.conf

18 Feb 12:21:52 - info: browser stderr: [1078:1078:0218/122152.302784:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared context.

18 Feb 12:21:53 - info: browser stderr: [1078:1259:0218/122153.419393:ERROR:object_proxy.cc(626)] Failed to call method: org.freedesktop.Notifications.GetCapabilities: object_path= /org/freedesktop/Notifications: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files

18 Feb 12:21:55 - info: browser stderr: ATTENTION: default value of option force_s3tc_enable overridden by environment.

18 Feb 12:21:55 - info: browser stderr: [1263:1263:0218/122155.386393:ERROR:sandbox_linux.cc(375)] InitializeSandbox() called with multiple threads in process gpu-process.

18 Feb 12:21:56 - info: Chrome loaded page /home/pi/piSignagePro/templates/1.html;
18 Feb 12:22:08 - info: {"LINK":"DOWN","DHCP":"","PING":"FAIL","DNS":"OK","NETSTATUS":"FAIL" }

18 Feb 12:22:08 - info: cpu serial number: 0000000063b87c57

18 Feb 12:22:08 - info: license status: true
18 Feb 12:22:08 - info: MAC Address : b8:27:eb:b8:7c:57  b8:27:eb:ed:29:02
18 Feb 12:22:08 - info: youtube-dl is available at /usr/local/bin/youtube-dl

18 Feb 12:22:09 - info: starting socketio to pisignage.tree-it.chwith ver: new
18 Feb 12:22:09 - error: socket.io: connect error event: xhr poll error
18 Feb 12:22:09 - info: socket.io works with 0.9.19
18 Feb 12:22:09 - info: starting socketio to pisignage.tree-it.chwith ver: 0.9.19
18 Feb 12:22:09 - info: socket.io: connected to server cNnvRcjiJ0Ng2OJOBhnjwith ver: 0.9.19
18 Feb 12:22:10 - info: Chrome loaded page /home/pi/piSignagePro/templates/1.html;?1550488930499
```

And I'm sure the HAT service as been killed on this log (5 lines before the end):
```
18 Feb 12:22:09 - error: socket.io: connect error event: xhr poll error
```

colloqi, 2019-02-18:
Few suggestions
- we do kill omx player using this command 'sudo kill $(pidof python omx.py)' you could see whether that is causing the issue?
- as you wrote, if the picofssd is started after pisignage, it works fine. Add in the start.sh a delay and then stop&start the service. 
- we have installations using RTC module and that should not be the issue
- to avoid xhr poll error, you could change the server address as "http://pisignage.tree-it.ch" but I don't think that could be the issue

dexluce, 2019-02-18:
> * we do kill omx player using this command 'sudo kill $(pidof python omx.py)' you could see whether that is causing the issue?

Reading the /home/pi/start.sh I thought instantly about this, but when I run the /home/pi/start.sh like this, the issue percist:
```
#!/bin/bash

unclutter -idle 5 -root &
cd /home/pi/piSignagePro

. /home/pi/.bash_profile
export WEBKIT_DISABLE_TBS=1
node pi-monitor.js
#sleep 10 
#sudo kill $(pidof python omx.py)
#sudo pkill omxplayer
#sleep infinity
```

> * as you wrote, if the picofssd is started after pisignage, it works fine. Add in the start.sh a delay and then stop&start the service.

That works. Notice, I had to add a "&" after your "node pi-monitor.js". Otherwise the script was blocked at "node pi-monitor.js". This means that your following lines never run right? :
```
sleep 10
sudo kill $(pidof python omx.py)
sudo pkill omxplayer
```

In the end, my working code for start.sh is:
```
#!/bin/bash

unclutter -idle 5 -root &
cd /home/pi/piSignagePro

. /home/pi/.bash_profile
export WEBKIT_DISABLE_TBS=1
node pi-monitor.js &
sleep 10
sudo kill $(pidof python omx.py)
sudo pkill omxplayer

### HOT-FIX: RESTART PICO HAT SERVICE AS IT IS SIGTERM ON "node pi-monitor.js"
### SEE: https://github.com/colloqi/piSignage/issues/73
sleep 60
sudo service picofssd restart
###

sleep infinity
```
This is not great, but it should be fine for our use.
It is interesting to notice that I had to do a 60s sleep before restarting the service. If I tried to start the service beetwin the time "pi-monitor.js" start, display the welcome screen, and launch the playlist, it won't start.

> * to avoid xhr poll error, you could change the server address as "http://pisignage.tree-it.ch" but I don't think that could be the issue

Indeed, it wasn't that.

colloqi, 2019-02-19:
Thanks, yes pi-monitor is a blocking call and we kill omx inside node.js process. I guess something in the beginning of start up process is creating the issue and hence the 60s delay you are talking about

closed 2020-07-30 by colloqi