colloqi/piSignage #74: Player name & hostname

zifnabbe, 2019-02-17:
I'm setting up a full new test playground for issue: player: https://github.com/colloqi/piSignage/issues/68.
I've the problem immediately from the first start. It played one day, but it didn't started up today.

So I've not sorted out what is really causing it... (I've not set tickers yet and I experience already the issue)
So, I'm looking at everything :-)

I did change my hostname: sudo hostnamectl set-hostname pi-X

I also noticed that the player name  (this is my test player) was placed in the hosts file as 
127.0.1.1 this-is-my-test-player

Is this by intention? [this was not the case with my previous player]

colloqi, 2019-02-18:
In 2.3.1 we added support to set up hostname which has few restrictions and hence the name change https://stackoverflow.com/questions/3523028/valid-characters-of-a-hostname

If you find the issue #68 I would like to see that player log

zifnabbe, 2019-02-18:
Well, after adjusting the hostname in the hosts file (to pi-X), the player starts when scheduled.
I'm now going to work out some new tests in the hope to find out about #68. Will let you know when I could trigger it (or not)...

closed 2020-07-30 by colloqi