colloqi/piSignage #75: warn: watchdog Timeout expired, killing video process, during playlist TV_OFF

zifnabbe, 2019-02-18:
In forever_out.log, I see the following warnings.
What is the meaning and why do these appear when the playlist isn't playing?

17 Feb 22:31:00 - info: Schedule is changed to playlist TV_OFF
17 Feb 22:31:00 - info:  **** turning tv off *****
17 Feb 22:31:30 - info:  **** tvservice off *****
17 Feb 22:55:50 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
17 Feb 23:29:35 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 00:03:20 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 00:37:05 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 01:10:51 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 01:44:36 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 02:18:21 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 02:52:06 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 03:25:51 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 03:59:36 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 04:33:21 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 05:07:07 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 05:40:52 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 06:14:37 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 06:48:22 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 07:22:07 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 07:55:52 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 08:29:37 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 09:03:22 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000
18 Feb 09:37:07 - warn: watchdog Timeout expired, killing video process: Choosin
g Wisely.mp4,258000

piSignage v2.3.1 on Raspberry Pi 3, Model B+

colloqi, 2019-02-18:
2 things here,
- the video file is greater than the duration (258 seconds) and hence the video process is killed. 
- Although the playlist has stopped, video may be playing internally - need to check this and correct the issue (thanks for reporting)

zifnabbe, 2019-02-18:
okay, the video is 265sec and time is set to 248sec

colloqi, 2020-01-05:
Was fixed in 2.4.0

closed 2020-01-05 by colloqi