colloqi/piSignage #76: [question] how to interpret Reports overview?

zifnabbe, 2019-02-18:
Hi,

How to interpret the following table?

[image]

What does '1' and '0' mean?
I don't understand the difference in '1' till 18/2 08:00AM and '0' from 18/2 09:00 AM

The player isn't playing since 10:30 PM (17/2), should start again at 11:00 AM (18/2).  
So why the difference?

Thanks in advance,

Regards, 
Tom.

colloqi, 2019-02-18:
Hi

Server-connected means player was in communication with server and playerUptime means player was "ON". 1 means true and 0 means false.

There are other events/reports which report about playlist play time

zifnabbe, 2019-02-18:
yes, but I don't understand the reason why the player was 'off' at 9 and 10 AM, while it all hours before it was 'on'.  
The player was not scheduled from 10:30 PM till 11.00 AM. So I would thought its was 'off' from 10:30 PM till 11.00 AM, and from then 'on'. So why was the player of from '9' AM and not before?