colloqi/piSignage #77: client access on personnal server with apache reverse proxy

gurvansz, 2019-02-27:
Dears,

I have a problem between my client (raspberry) and my server how use a apache reverse proxy for the node.js application.
My server address is : http://website.com/node
the reverse proxy start application when a query is done to this URL by calling http://website.com:3000
The server run perfectly but when I want to add my player, there is no communication between the two parts. I change the server address with lot of example on the raspberry webpage but nothing work.
Could you help me?
Thanks a lot

colloqi, 2019-02-27:
How do you access the server UI? use the same address as server address in pi-player configuration and it should work. For e.g. if you access the server UI as http://website.com/node give the address as server address in player

gurvansz, 2019-02-27:
Thanks for your reply.
I tried to use the exact same address, but I have an error on the rapsberry logs :

> error: socket.io: connect error event: xhr poll error

colloqi, 2019-02-27:
1. can you ping the address http://website.com/node, otherwise there could be DNS issues?
2. If ping goes through, try to see the apache reverse-proxy logs to see what happens when raspberry "websocket" request comes

gurvansz, 2019-02-27:
1. ping is ok
2. i made a tail on apache logs and this is when client try to communicate with server :

> [27/Feb/2019:15:00:04 +0800] "GET /newsocket.io/?EIO=3&transport=polling&t=MajmCtK&b64=1 HTTP/1.1" 404 473 "-" "node-XMLHttpRequest"

I fill the apache doesn't redirect the websocket to the good path. My server URL is http://website.com/node so why he try to redirect to direct /nwesocket.io and not to /node/newsocket.io ?
I changed my apache reverse proxy configuration file with rewrite rules for it but i feel it's not ok...

gurvansz, 2019-02-27:
For information, this is my apache configuration file for reverse proxy :
`<VirtualHost *:80>
	ServerName website.com
	ServerAlias website.com

	ServerAdmin webmaster@localhost

	ErrorLog ${APACHE_LOG_DIR}/error.log
	CustomLog ${APACHE_LOG_DIR}/access.log combined
	
	#configuration for node.js service
	ProxyPreserveHost On
	ProxyRequests Off

	ProxyPass /node http://localhost:3000
	ProxyPassReverse /node http://localhost:3000

	#configuration for socket.io service raspberry
	RewriteEngine On
    RewriteCond %{REQUEST_URI}  ^/newsocket.io            [NC]
    RewriteCond %{QUERY_STRING} transport=websocket    [NC]
    RewriteRule "/(.*)"        ws://localhost:3000/$1 [P,L]

</VirtualHost>`

The frontend of the server works well.

colloqi, 2019-02-27:
I guess if you can, modify your server url to http://node.website.com/  and work on configuration file etc. it may work. I guess socket.io.client is not appending "/node" for requests.

Other option is add ProxyPass for /newsocket.io to http://localhost:3000/newsocket.io (not sure of this as I do not know Apache)

gurvansz, 2019-02-27:
I prefer to use second option for the moment.
I think I have maybe a good direction.
I added the Proxypass setting and now on the server log, I have 401 error and not 404 error when newsocket.io is requested.
Need to tweak again some conf files...

gurvansz, 2019-02-28:
Problem came from the server provider. He closed my port and raspberry can't communicate without it.
Now it's ok.
Thanks for your time

closed 2019-02-28 by gurvansz