colloqi/piSignage #80: [Suggestion] Schedule Playlist allow cron as plain scheduling

fdsoinfsdifidncoiddcn, 2019-03-22:
CRON typed scheduling for playlists would add a lot comfort. My customers need scheduling every two weeks from monday to thursday. This is not possible right now so we need to do this by hand every month.

hitesh83, 2019-03-22:
This is very much possible in current setup, total 30 playlist with 29 CRON type scheduling playlist can be added.
[image]

fdsoinfsdifidncoiddcn, 2019-03-22:
thanks for the answer. I know this panel but I have no idea how to schedule a playlist every 2 weeks from wednesday to friday. Can you help me with that?

hitesh83, 2019-03-22:
surely will help you,
please post detail exactly how you want to setup your playlists.

fdsoinfsdifidncoiddcn, 2019-03-22:
I need a playlist that runs biweekly from wednesday-friday

hitesh83, 2019-03-22:
So there will be playlist -1 running Wednesday to Friday that too after every 2nd week or something like that.
And playlist-2 running rest of the days.

Are you running open source or using pisignage.com services ?

Mail me detail playlist arrangements you want.