colloqi/piSignage #81: tvservice is switched off a few seconds after startup

jonathanschneider, 2019-03-26:
I've noticed that in certain situations after booting the Pi tvservice is switched off after a few seconds. The server states that the intended playlist is still active and "mpv" for example is listed when executing "ps" but the TV is showing "no signal". Executing "tvservice -p" solves the issue.

Steps to reproduce the issue:
1. Set default playlist to TV_OFF and deploy.
2. Pull power supply from Pi.
3. Set default playlist to anything but TV_OFF and deploy.
4. Power on Pi.

Pi will boot up, pull new configuration from server and will play playlist for about 10 seconds. Then TV will show "no signal" and executing "tvservice -s" will output "state 0x120002 [TV is off]".

Excerpt from "forever_out.log":

26 Mar 21:02:34 - info: wget stopped with code 0 and signal null; downloadInProgressCount: 1
26 Mar 21:02:34 - info: COPIED SERVER FILES Test SUCCESSFULLY.
26 Mar 21:02:34 - info:  **** turning tv on *****
26 Mar 21:02:35 - info: openvg_display arguments: 1 0 0
26 Mar 21:02:35 - info: *** Changing Playlist *** to Livestream
26 Mar 21:02:35 - info: Change playlist,deleting files: Test.tmp,_logs,_thumbnails
26 Mar 21:02:35 - info: browser stderr: [894:894:0326/210235.827507:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared context.

26 Mar 21:02:37 - info: browser stderr: [894:1184:0326/210237.135643:ERROR:object_proxy.cc(626)] Failed to call method: org.freedesktop.Notifications.GetCapabilities: object_path= /org/freedesktop/Notifications: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Notifications was not provided by any .service files

26 Mar 21:02:39 - info: browser stderr: ATTENTION: default value of option force_s3tc_enable overridden by environment.

26 Mar 21:02:39 - info: browser stderr: [1188:1188:0326/210239.278300:ERROR:sandbox_linux.cc(375)] InitializeSandbox() called with multiple threads in process gpu-process.

26 Mar 21:02:40 - info: Chrome loaded page /home/pi/piSignagePro/templates/1.html;
26 Mar 21:02:42 - info: Chrome loaded page /home/pi/piSignagePro/templates/1.html;?1553630562190
26 Mar 21:03:00 - info:  **** tvservice off *****
26 Mar 21:03:05 - info:  **** cec-client on *****
26 Mar 21:03:05 - error: Already playing started

colloqi, 2019-03-28:
Hi, This is because the config update is yet to update the schedules in player, so typically with the next config update which typically happens every 5 minutes (configurable under settings) the player should come on and play the new playlist. Always pressing deploy updates the player instantly if it is online

jonathanschneider, 2019-03-28:
Would it be possible that the player also updates its schedule after it booted and pulled the new configuration from the server instead of waiting for the next update cycle?

colloqi, 2019-03-28:
It does, we will see why it is missing poweron update of configuration and fix if possible

colloqi, 2020-01-05:
I guess it is not an issue in recent releases

closed 2020-01-05 by colloqi