colloqi/piSignage #82: No sound if HDMI cable isn't connected during boot

jonathanschneider, 2019-04-09:
I have a player that is being set up and taken down every week. Sometimes the player would not output any sound and the only remedy I found was to restart it. I did not report the issue right away because I couldn't reproduce the issue but I finally figured out that the problem occurs when the HDMI cable isn't connected when the player boots.
I know how to avoid the issue now but would it be possible that whatever is checked or set during boot is also set whenever a playlist starts?

Thanks in advance!

colloqi, 2019-04-09:
Hi, Are you using a standard resolution TV?

jonathanschneider, 2019-04-09:
Yes!

colloqi, 2019-04-09:
Can you change the values of hdmi_mode and hdmi_group  in /boot/config.txt to suit your TV? more details at https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

jonathanschneider, 2019-04-09:
hdmi_group was already set to 1 and I changed hdmi_mode to 19 (720p, 50 Hz) but that didn't make a difference.
Realising that it has nothing to do with piSignage but the Raspberry Pi, I set the audio output to HDMI instead of auto with this command:

`amixer cset numid=3 2`

Now audio always works even if the HDMI cable is connected after the Pi booted.

Thank you for the support. The issue can be closed.

closed 2019-04-09 by jonathanschneider

colloqi, 2019-04-10:
👍