colloqi/piSignage #85: Advert. Playlist not playing durring Video Playlist

PhoenixZeroDown, 2019-06-28:
Hi there,
maybe this function is on purpose, but our adv. Playlist won't play during a video playlist. After this playlist has finished the adv. playlist starts. With RSS playlist or picture slideshow it is working fine. Do I miss something here? 
For this test there is just the video playlist set as default group playlist and the adv. playlist as additional playlist.
Help is much appreciated here.

colloqi, 2019-06-28:
Hi,
Under Group settings you have a setting "Break VIdeo" to show adverts. Use the same.

PhoenixZeroDown, 2019-07-03:
Thank you for your quick response, but unfortunately couldn't get it to work. 
Another thing, that isn't working is, when you have playlist inside the side zone of the main playlist. 

My Setup:

Main Playlist is Layout 2b (left main zone is playing multiple pictures, right side zone is playing another playlist). 

Another Playlist is Layout 1 (contains two assets, one is weblink to a wordpress page, and the other one is just a text message).

Advt. Playlist is Layout 2b (left main zone dummy text message, right side zone is just a text message)
Settings are: Make this as Advt. Playlist and Play only in side/banner zones.

If set up like this, no Advt. Playlist is playing. If set up to play only in the main zone it is working.

Interesting thing is, Domination Playlist works, but of course, shows assets in both zones.

Is this some kind of bug or am I missing something?

colloqi, 2020-07-30:
Although it is late :), independent playlists can not be controlled from main playlist

closed 2020-07-30 by colloqi