colloqi/piSignage #88: Video duration is not respected when duration is shorter than video length

Luke1982, 2019-07-17:
When I set a video's duration shorter than the actual video length, the video will still just play the full length. In the JSON file on the player (the JSON file that represents the playlist), the duration is set like I set in on the server. Still, the player does not respect the shorter duration.

colloqi, 2019-07-18:
Just to take care that video exit event is honoured before timeout, we add 10 seconds to duration and put the timeout. In this case it matches 15 seconds and matches the video duration!

Luke1982, 2019-07-19:
Hmm, let me test that some more since I've seen it run much longer than 10 extra seconds...

bbp-estu, 2019-07-25:
I'm also facing same issue. I have uploaded a mp4 video of 219 seconds. In the playlist, duration is set to 15 sec. But still videos plays until end (whole 219 sec)

bbp-estu, 2019-07-26:
Update: upgraded software to 2.5.0 (from 2.4.1) and the issue is solved

Luke1982, 2019-07-26:
What is the publishing date of 2.5.0? I believe I'm still on 2.4.1, so this might also solve my issue. If it does I can close the issue.

bbp-estu, 2019-07-27:
I'm not sure about published date,maybe recently. You can check under registered players section the version of pisignage and if upgrade available there should be an icon/button to the left of 'terminal' button.
[image]

closed 2020-01-05 by colloqi