colloqi/piSignage #89: <div>Layer above Video area possible

PhoenixZeroDown, 2019-07-24:
Hi there,

is it possible to place a <div> Layer partially over the video area? I'm building a custom layout with some overlapping DIVs and with only pictures in the main zone it works flawless, but the video is above my DIVs so that my layout kinda breaks. My assumption of how the video playback works is, that it just puts the video over the template.

Any help is much appreciated.

colloqi, 2019-07-26:
Hi

Under layout popup, you have second, third and fourth tabs wherein you can specify the video window size

regards

sk-69, 2019-08-29:
Hi,
I have the same problem. I would like to display a PNG image above a video but I can't. 
Maybe we have to add a z-index property ?
Could you help me please ?
Thanks :-)

PhoenixZeroDown, 2019-08-29:
Adding z-index property won't help in the case of a video because the video playback window is set above the html site, so setting z-index won't work here.

sk-69, 2019-08-29:
Thanks @PhoenixZeroDown 👍  
Do you have an other solution ?

PhoenixZeroDown, 2019-08-29:
Not quite a solution, but you could create a single html file with an embedded mp4 video file, create an asset with an url to it an add it to your playlist. Performance of the video playback is an issue here because there is no hardware acceleration, at least with my Raspberry Pi 3b it was quite slow.

closed 2020-01-05 by colloqi