colloqi/piSignage #90: Will not boot when set to portrait on RPI4

bwlange3, 2019-08-17:
For some reason when you change the orientation to portrait mode (no matter weather you have a specific resolution set or its set to auto).  I know there is a recommendation for using landscape mode, but I thought I would share that portrait is not working at all.  Hopefully support for this is soon to come.

interdif, 2019-11-03:
same problem ...

interdif, 2019-12-02:
any idea for a fix ?

colloqi, 2019-12-03:
With 2.6.1 image, it should boot, only issue is the hdmi_group and hdmi_mode is set to 1 & 16 (1080p) and legacy mode driver will be used (not fkms). You could change the values in /boot/config.txt for hdmi_group & hdmi_mode to suit your display and display should appear

interdif, 2019-12-03:
Hi,

It can work with 4k display?

(Only html and img display no video need)

colloqi, 2019-12-03:
Yes, for portrait mode you may have to program the hdmi_mode value accordingly. use the following in piShell to get the values for your display
tvservice -m DMT
tvservice -m CEA

interdif, 2019-12-03:
OK thx, I will try that tomorrow

One last question, it is possible to use both hdmi as clone?

colloqi, 2019-12-03:
For non-video assets, yes

interdif, 2019-12-03:
Do you know the config options for that?

colloqi, 2019-12-03:
If both cables are connected on poweron, it should just show up at first screens resolution.

interdif, 2019-12-03:
OK I try that asap, thx a lot

interdif, 2019-12-18:
Hi,

Tried to set portrait and set the resolution manually like you said with
the parameters found with the two commands. Sadly it doesn't work, I get a
band of image of 10% the rest is black....

interdif, 2019-12-27:
Hi,

any idea ??

colloqi, 2020-01-05:
Can you let us know the player and installation when it is online  at support at pisignage.com? we will see the config.txt options?

interdif, 2020-01-05:
You mean the player ID? I can let it online but I need to know the config
you want, défaut with portrait so in 1080 or the 4k with default on screen

colloqi, 2020-01-05:
One of these options should work
- Default with 4K in LANDSCAPE mode and use 2ap or 2bp for layout under playlists (this will make it portrait)
- 1080p resolution and portrait mode under group settings

interdif, 2020-01-05:
i have tested with  default 4K and changed the layout, it does work for txt and img but not working for html link ... and i need it, also i ave tested PDF and it show a very non center display

colloqi, 2020-01-07:
Can you send us the screenshot and account info at support at pisignage.com?

interdif, 2020-01-07:
ok, i will do it today 

if necessary i can give you a ssh

interdif, 2020-01-07:
i found something, whne i set a webpage with "weblink" with layout 2ap/2bp it dont show in portrait, but when its with "link" it work, problem is i cant set auto scroll and other options then ...

interdif, 2020-01-07:
and "link" doesnt work with all web pages, some i have tested dont show anyting except black or wite page

colloqi, 2020-01-08:
weblink needs hardware portrait setting (under Group) to work with 2ap/2bp. link type uses iframe and it does not work for CORS links.
We will raise an internal issue to make weblink work without hardware portrait setting

closed 2020-07-30 by colloqi