colloqi/piSignage #91: CRC Error on RPI4 Player Image

PhoenixZeroDown, 2019-09-02:
Hello,
got a bunch of new RPI4s and wanted to install the Player Image, but when I want to extract the ZIP file I always get an CRC Error. Tried 7-zip and the Win10 internal zip tool. I also tried both download links, but didn't made any difference.
Can someone help?

maroliict, 2019-10-11:
Did you find a solution? I get similar issues. Can download, but cannot extract and for sure cannot put it on a micro SD card

PhoenixZeroDown, 2019-10-11:
No. But I downloaded the uncompressed image from the link below, the same as it is mentioned in the instructions for the pi player:
[Uncompressed Image](https://drive.google.com/open?id=1UJBc_tRS07bi8U_ytLTS5bzLl3aVbdzT)

maroliict, 2019-10-11:
Thanks. Will try that now...

closed 2020-01-05 by colloqi