colloqi/piSignage #94: YouTube Live issue

hitesh83, 2019-10-03:
With current full image 2.4.1 (fresh install), and older version below that do not work with YouTube live. 
This  **error** for YouTube Live  -- >

```
3 Oct 10:47:58 - info: livestreamer/ydl error ERROR: 21X5lGlDOfg: YouTube said: This video is unavailable.
3 Oct 10:48:00 - info: livestreamer/ydl error WARNING: Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.

```

This **error** for YouTube normal video  -- >
```

3 Oct 10:52:11 - info: livestreamer/ydl error ERROR: LNKhpRwdEBs: "token" parameter not in video info for unknown reason; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.

3 Oct 10:52:13 - info: livestreamer/ydl error WARNING: Assuming --restrict-filenames since file system encoding cannot encode all characters. Set the LC_ALL environment variable to fix this.

```

With Manual update of YouTube-dl by below command only Youtube normal video works with its full length. 
But YouTube live restarts feed every 5 sec.

```
sudo curl -L https://yt-dl.org/downloads/latest/youtube-dl -o /usr/local/bin/youtube-dl
sudo chmod a+rx /usr/local/bin/youtube-dl
```

**This is not a PiSignage issue, but actually youtube-dl issue**

hitesh83, 2019-10-04:
**Update -** 

just a small workaround for **YouTube** using browser, which is already available in Player

1- go to Youtube - go to actual video or live feed
2- click on embed - un-tick  _"Show player controls"._
3- copy the code and paste it in HTML
4- add _&mute=1&autoplay=1_ to the youtube url in given code [in <iframe src=... >]

full e.g.   

_<iframe width="320" height="320" src="https://www.youtube.com/embed/XBPjVzSoepo?controls=0&mute=1&autoplay=1" frameborder="0" allow="accelerometer; autoplay; encrypted-media; gyroscope; picture-in-picture" allowfullscreen></iframe>_

here you can customise width, height as per requirement. change mute=0 for audio.
Youtube link used here is NASA live feed, you need to change it to your desire one.

now make a html file put all margins to zero, and put only iframe inside html tag
upload this html page to pisignage server and schedule it on your player.

For one secound while starting the feed, you will see youtube logo on bottom, and video name on top, after 1 sec it will run without any text or logo etc..

Hope to get youtube-dl working again, so deploying youtube link will be more easy again.

colloqi, 2020-01-05:
Whenever youtube is not working, please follow the following
1. Upgrade youtube-dl to the latest with piShell or ssh command on player, "sudo -H pip install --upgrade youtube-dl"
2. piSignage player is not caching youtube link so sometimes youtube may block the access (please check tail -200 /home/pi/forever_out.log for the error messages)
3. Use the share  link of Youtube rather than page URL
4. Use the mpv player option under Group settings

closed 2020-01-05 by colloqi